NVIDIA-AI-IOT/jetbot #407: Setuping the Jetson Nano 4gb for JetBot

user19981, 2021-03-24:
Hi
Every time i start up jetson nano, it displays a black screen with the same error line appearing again and again every 10 seconds:
tegra-i2c 7000c400.i2c: no acknowledge from address 0x3c

I try to login but I'm not even sure of what the login is because it's a device i got from an older project. However I know the password because i had to choose it.

The last thing i did is following the different steps from: https://jetbot.org/v0.4.3/software_setup/docker.html

I'm not sure but I think the problem comes from the command line: ./scripts/configure_jetson.sh 

Any advices on what I should do to fix this? 
How can I start over just like the Jetson Nano was new?

[image]

tjp1940, 2021-03-24:
I had that same problem and discovered that there was a loose connection
with the display and it was not powered up.  If any of the i2c devices like
the motor controller, display, or wifi is not functioning properly the
jetbot wants it fixed before it continues to boot up.  That is my opinion
as I am not an expert by far.  Anyway I recommend checking all the jetbot
connections to assure nothing is miswired.
TJP

tokk-nv, 2021-03-30:
Hi user19981,

I understand the error messages are worrying, but you should still be able to login by just starting typing your username.

If the image is JetBot image, you want to try `jetbot` for username and `jetbot` for password.

OZtwo, 2021-04-29:
I'm at this stage as well. They are VERY unclear about how to install JetBot. From what I understand is you can not download and install the two images options shown first. Just above the two file options (2g/4g) they have this very odd notice:

**Attention
To use a sdcard image that is based on JetPack 4.5, you need to run through the initial ("oem-config") setup using the original JetPack 4.5 Jetson Nano SD card image (Nano 2GB, Nano (4GB)) first. (two links given)
This step will configure the bootloader on Jetson Nano hardware as needed for the latest SD card we released for JetBot.**

I'm stick at this point since there really is no easy way to move on to the next step. For this  you need two Linux systems and the SDK Manager on the host PC to install the needed package! WTF is going on here?

The first image they say you must download is only the OS and nothing else. to get JetBot working you must install the second (or I'm sorry, the main package) to get it working. I hope they fix this.

At the end, if you do download JetPack 4.5 and flash that image to the SD the system will boot, but will only boot to a blank screen asking for a log in with that error showing up every 10 seconds.

santaimpersonator, 2021-09-30:
@OZtwo You might want to try the [Nvidia forum](https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-nano/76) for more technical assistance.

### Error message

The error message is from a service running the background (it only represents that the service wasn't able to access the OLED display through an I2C connection). Your board is running fine and you aren't at a blank screen; you are actually at the login terminal: similar to this screen from the movie war games https://www.youtube.com/watch?v=uCWKZWieMSY&t=237s

The script `./scripts/configure_jetson.sh` disables the graphical desktop GUI and brings users to a terminal when the Jetson is rebooted.

1. Just type in the <username> and hit the `[ENTER]` key
2. Then type in the <password>and hit the `[ENTER]` key
    * As @tokk-nv mentioned, the defaults as `jetbot` for both
3. Once logged in, you should see something like this:
    [image]
4. Now, you can stop the background service or clear the output with the `clear` command.

However, if you already finished setting up the Jetbot package/Docker container with all the steps here: https://jetbot.org/v0.4.3/software_setup/docker.html
You just need to login to the Jetbot using the IP address from the WiFi setup: https://jetbot.org/v0.4.3/software_setup/wifi_setup.html#check-ip-address

### Using Jetpack 4.5:
I think you might have dove to deeply into the SDK and kernel documentation. You only need to flash the SD card and set up the WiFi connection to start using the Jupyter notebooks. The [instructions for the Docker container](https://jetbot.org/v0.4.3/software_setup/docker.html), are for users who want to update their current image to use the Docker containers (as noted at the top of the instructions).

***Note:** All the images are in `.zip` foders, you just need to extract them and [flash an SD card](https://jetbot.org/master/software_setup/sd_card.html#step-2-flash-jetbot-image-onto-sd-card)

1. Flash the blank/plain image onto an SD card **(see note above)*
    * Listed in the `ATTENTION` call out box of **Step 1**:
    [image]
2. Run through the `oem-config` tool (runs automatically on boot) to flash the new firmware onto the Jetson (the last step)
    * This updates the firmware on the Jetson and you can't undo this step as noted in the pop-up dialog box. (i.e. Essentially, you won't be able to use older images from Jetpack 4.2 and earlier)
3. Shutdown the Jetson
4. Flash the pre-built image onto an SD card **(see note above)*
    * Listed at the top of **Step 1**:
    [image]
5. Run through the rest of the setup steps