NVIDIA-AI-IOT/jetbot #410: Completely Headless setup

errolt, 2021-03-29:
I'm trying to set up the Jetbot image completely headless. The latest Jetson image allows headless setup over the serial port that the Jetson presents over USB, if I'm not mistaken.

The Jetbot Wifi setup page contains this:

> If you are using Jetson for the first time right after you flash your SD card, then you don't have this user account set up.
>In that case, you want to use the USB serial functinality, and use your serial terminal software on your PC to have accesss to the Jetson's serial terminal.

If I connect to the serial port that the Jetson hosts over USB then I just get to a login screen, and I don't have a login yet.
If I connect to the debug serial port on the pin header then I get boot info, and then a login screen.

Is there no way to set up without a screen/keyboard connected?

tokk-nv, 2021-03-30:
Hi errolt,

My understanding is that you are trying to do the setup from scratch based on the stock JetPack image. (not JetBot image)
And you are doing this without connecting the HDMI display.

> If I connect to the serial port that the Jetson hosts over USB then I just get to a login screen, and I don't have a login yet.
> If I connect to the debug serial port on the pin header then I get boot info, and then a login screen.

I think I have run into this issue before.
I did not get to find the cause of this, but this will persist, and you just need to reflash the SD card and try again.

Next time you boot with your fleshly flashed SD card, you want to not connect any HDMI display, and I tried plugging in the micro USB cable at the different time (I think plugging micro USB cable after 1-3 min after boot up gave me the best chance, but I'm not sure if I'm remembering things correctly...)

errolt, 2021-03-30:
Hi tokk-nv,

I was being dumb. I am using the JetBot image, but was expecting a script like on the JetPack image that ask you to create a user and set up etc.

For some reason I read through the Jetbot setup instructions many times, but every time I missed this:

> Log in using the user jetbot and password jetbot

Now I can log in over USB or over Ethernet to set up wifi. Screen and keyboard not required!

Sorry about logging dumb issue!

Thank you,
Errol

closed 2021-03-30 by errolt