NVIDIA-AI-IOT/jetbot #414: PiOLED is not working

Thenknr, 2021-04-02:
HI guys!!

I'm building a JETBOT with JETSON NANO 2GB and Yahboom Jetbot AI robot.

I'm following this URL 'https://github.com/NVIDIA-AI-IOT/jetbot/wiki/software-setup'

while I was following, at Step 3 - Connect JetBot to WiFi,  there was no problem until

Log in using the user jetbot and password jetbot.

Connect to a WiFi network using the Ubuntu desktop GUI

However piOLED shows nothing on step

'Your Jetson Nano should now automatically connect to the WiFi at boot and display its IP address on the piOLED display.'

Is there any solution to figure this out?

Although piOLED is not showing off, it's still connected to wifi and I checked with Terminal using 'ifconfig' command.

[image]

Srafington, 2021-04-09:
If you start the Jetson in headless mode (but still connected to a monitor), does the terminal constantly repeat an error about not getting a response from 0x3a?

NoahRCarver, 2021-09-25:
Hi, I'm having a similar problem (I'm using the waveshare jetbot kit however)

I recieved
`OS error: [Errno 121] Remote I/O error`

santaimpersonator, 2021-10-14:
@NoahRCarver Then you have a wiring/connection issue or you have the wrong I2C address in your code. The Error 121 means that the OLED display can't be reached through the I2C bus at the specified address.

This is the code that essentially runs for the OLED display: https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/jetbot/apps/stats.py
It [runs as a service](https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/scripts/create-sdcard-image-from-scratch.sh#L113-L118) in the [docker container](https://github.com/NVIDIA-AI-IOT/jetbot/tree/master/docker) *(see the `display` folder)*.

closed 2022-07-18 by jaybdub