NVIDIA-AI-IOT/jetbot #419: Jetbot is too slow Unable to run data collection code on it.

Pranav-India, 2021-04-03:
Hello,
I am trying to run the data collection code for the road following task using the given notebook but the jetbot computation is too slow and I am not able to run any cell I checked the stats of the cpu-gpu usage using the jetson package below is the output I received can you tell me is there something I have done wrong or am I missing something? 

[image]

Thank you.

Srafington, 2021-04-09:
It looks like you are configured in low power mode (5W), you'll want to switch to high power mode by running 
sudo nvpmodel -m 0

Pranav-India, 2021-04-19:
Thank you I did that , it solved the issue..

closed 2021-04-19 by Pranav-India