NVIDIA-AI-IOT/jetbot #420: IndexError: tuple index out of range (road follower data collection notebook)

Pranav-India, 2021-04-03:
I am running road follower data collection notebook.
`widgets.jsdlink((controller.axes[2], 'value'), (x_slider, 'value'))`
`widgets.jsdlink((controller.axes[3], 'value'), (y_slider, 'value'))`

This line is causing the following error.

`
IndexError                                Traceback (most recent call last)
<ipython-input-8-68c93abe2934> in <module>
----> 1 widgets.jsdlink((controller.axes[2], 'value'), (x_slider, 'value'))
        2 widgets.jsdlink((controller.axes[3], 'value'), (y_slider, 'value'))

IndexError: tuple index out of range
` 
How can I remove this error?

Srafington, 2021-04-09:
You need to have a connected controller (like a gamepad)

tranvietlong, 2021-04-17:
in the previous step :
controller = widgets.Controller(index=0)

display(controller)

Pranav-India, 2021-04-18:
Hi thank you for the responses I got the gamepad and now it is working fine, I did the data collation and tried to train the model but I am getting following error
`OSError: Caught OSError in DataLoader worker process 2.`
`Original Traceback (most recent call last):`
 `File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop`
    `data = fetcher.fetch(index)`
  `File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch`
    `data = [self.dataset[idx] for idx in possibly_batched_index]`
  `File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>`
    `data = [self.dataset[idx] for idx in possibly_batched_index]`
  `File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataset.py", line 257, in __getitem__`
    `return self.dataset[self.indices[idx]]`
  `File "<ipython-input-2-3df4b79a2136>", line 23, in __getitem__`
    `image = PIL.Image.open(image_path)`
  `File "/usr/lib/python3/dist-packages/PIL/Image.py", line 2590, in open`
    `% (filename if filename else fp))`
`OSError: cannot identify image file 'dataset_xy/xy_140_125_cdde6f46-a024-11eb-9caa-00044be7083b.jpg' `

closed 2022-07-15 by jaybdub