NVIDIA-AI-IOT/jetbot #423: Upgrade Jetbot PyTorch and Torchvision version

abuelgasimsaadeldin, 2021-04-12:
Hi, I noticed that Jetbot is using **torch version 1.3.0** and **torchvision version 0.4.0** I would like to know has anyone upgraded the torch and torchvision versions for PyTorch? I would like to know if there are any problems that could be faced in the scripts by doing so.

The reason would like to upgrade is because when training my models on my GPU desktop which has a **torch version 1.8.1** and **torchvision version 0.9.1** and try to load it the model on the Jetbot I receive an error which I believe is caused due to the old torch version.

This is the error I receive when trying to load the model:

```
RuntimeError: version_number <= kMaxSupportedFileFormatVersion INTERNAL ASSERT FAILED at ../caffe2/serialize/inline_container.cc:131, please report a bug to PyTorch. Attempted to read a PyTorch file with version 3, but the maximum supported version for reading is 1. Your PyTorch installation may be too old. (init at ../caffe2/serialize/inline_container.cc:131)
frame #0: c10::Error::Error(c10::SourceLocation, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) + 0x78 (0x7f654678d8 in /usr/local/lib/python3.6/dist-packages/torch/lib/libc10.so)
frame #1: caffe2::serialize::PyTorchStreamReader::init() + 0x1a9c (0x7f311b5444 in /usr/local/lib/python3.6/dist-packages/torch/lib/libtorch.so)
... continued
```

tokk-nv, 2021-04-21:
The latest jetbot images (or containers) are now based on `nvcr.io/nvidia/l4t-pytorch:r32.4.3-pth1.6-py3` container image.
So it should have PyTorch v1.6 - torchvision v0.7.0 combination.

http://jetbot.org/v0.4.3/software_setup/sd_card.html#latest-release

To use PyTorch 1.8.1 with Torchvision 0.9.1, you may want to try rebuilding the base container with PyTorch v1.8 - torchvision v0.9.0.
https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-8-0-now-available/72048
Unfortunatelly, we don't have the PyTorch 1.8 containers for JetPack 4.5 yet.
https://ngc.nvidia.com/catalog/containers/nvidia:l4t-pytorch/tags

abuelgasimsaadeldin, 2021-04-22:
Hi @tokk-nv, thank you for the reply, I actually am using a Jetbot from Waveshare and if not mistaken the image from the Nvidia AI IOT Jetbot would not function properly when using the Waveshare Jetbot (correct me if I'm wrong).

I'm not exactly sure on what version of JetPack the Waveshare Jetbot Image is based on but if I'm not mistaken it has not been updated for awhile. This is the latest image found on their software setup page:

https://drive.google.com/file/d/1G5nw0o3Q6E08xZM99ZfzQAe7-qAXxzHN/view

closed 2022-07-15 by jaybdub