NVIDIA-AI-IOT/jetbot #424: ZMQ camera container: Failed to start camera publisher

Hoehlenbaer, 2021-04-13:
Hi together,
I pulled the latest image from docker-hub and run it.
Unfortunately the publisher fails to start the camera and the therefore I only get black frames.
Docker log:

starting camera
nvbuf_utils: Could not get EGL display connection
(Argus) Error FileOperationFailed: Connecting to nvargus-daemon failed: No such file or directory (in src/rpc/socket/client/SocketClientDispatch.cpp, function openSocketConnection(), line 201)
(Argus) Error FileOperationFailed: Cannot create camera provider (in src/rpc/socket/client/SocketClientDispatch.cpp, function createCameraProvider(), line 102)
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:639 Failed to create CameraProvider
(python3:6): GStreamer-CRITICAL **: 14:36:05.080: gst_mini_object_set_qdata: assertion 'object != NULL' failed
0
nvarguscamerasrc sensor-mode=0 ! video/x-raw(memory:NVMM), width=816, height=616, format=(string)NV12, framerate=(fraction)21/1 ! nvvidconv ! video/x-raw, width=(int)224, height=(int)224, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink name=sink
starting camera
nvbuf_utils: Could not get EGL display connection 

This error repeats a couple of times. 
When starting the camera publisher script in a jupyterhub terminal, everything works fine...

jaybdub, 2021-04-21:
Hi @Hoehlenbaer ,

Thanks for reaching out!

Do you mind sharing the commands you used to run the docker container?

Best,
John

Hoehlenbaer, 2021-04-23:
Hi jaybdub,
Actually I only issued a simple "sudo docker run <imageID>" command...
However I realized the container is not running at all. It starts but immediately exits with error code 0...

I thought there might be a problem with the script, so I started the container again, but removed the command. It still exits with error code 0. No idea how I can debug this, if it's refusing to run.

Regards
Marco

Hoehlenbaer, 2021-04-23:
Just to understand: The camera container version needs to match SD card image version, isn't it? So for SD card with 32.5.1, it needs to be camera with 32.5.0...

I'm still in the very steep part of my jetbot learning curve :-(

Hoehlenbaer, 2021-05-06:
Hi jaybdub,

I now did a commit of the container to a temporary_image and run this with 
--entrypoint=bash -it.

The container runs interactively. Then I tried to start the zmq_camera_publisher.py script. This creates the same errors as written above. Running the same python script inside the jupyterlab container works without issues.

Can I somehow check the status of the nvargus-daemon?

Regards 
Marco

Hoehlenbaer, 2021-05-07:
Hi together,
I bashed into the camera container.
cd /dev and ls gives this result:

root@de32b040da80:/dev# ls
console  mqueue           nvhost-dbg-gpu  nvhost-prof-gpu  random  tegra_dc_0     zero
fb0      null             nvhost-gpu      nvhost-vic       shm     tegra_dc_1
fb1      nvhost-as-gpu    nvhost-msenc    nvmap            stderr  tegra_dc_ctrl
fd       nvhost-ctrl      nvhost-nvdec    ptmx             stdin   tty
full     nvhost-ctrl-gpu  nvhost-nvjpg    pts              stdout  urandom
root@de32b040da80:/dev#

Shouldn't there be much more devices? At least "video0" should be there to use a camera, shouldn't it?

Regards
Marco

Hoehlenbaer, 2021-05-07:
Cross checked with my jupyter container. Looks much different:

root@nano-4gb-jp45:/dev# ls
autofs                  loop4               nvhost-ctrl-isp.1  quadd              tty1   tty46    ttypa
btrfs-control           loop5               nvhost-ctrl-nvdec  quadd_auth         tty10  tty47    ttypb
bus                     loop6               nvhost-ctrl-vi     ram0               tty11  tty48    ttypc
console                 loop7               nvhost-ctxsw-gpu   ram1               tty12  tty49    ttypd
constraint_cpu_freq     mapper              nvhost-dbg-gpu     ram10              tty13  tty5     ttype
constraint_gpu_freq     max_cpu_power       nvhost-gpu         ram11              tty14  tty50    ttypf
constraint_online_cpus  max_gpu_power       nvhost-isp         ram12              tty15  tty51    uhid
cpu_dma_latency         max_online_cpus     nvhost-isp.1       ram13              tty16  tty52    uinput
cpu_freq_max            media0              nvhost-msenc       ram14              tty17  tty53    urandom
cpu_freq_min            mem                 nvhost-nvdec       ram15              tty18  tty54    v4l-subdev0
cuse                    memory_bandwidth    nvhost-nvjpg       ram2               tty19  tty55    v4l-subdev1
emc_freq_min            min_online_cpus     nvhost-prof-gpu    ram3               tty2   tty56    vcs
fb0                     mmcblk0             nvhost-sched-gpu   ram4               tty20  tty57    vcs1
fb1                     mmcblk0p1           nvhost-tsec        ram5               tty21  tty58    vcs2
fd                      mmcblk0p10          nvhost-tsecb       ram6               tty22  tty59    vcs3
full                    mmcblk0p11          nvhost-tsg-gpu     ram7               tty23  tty6     vcs4
fuse                    mmcblk0p12          nvhost-vi          ram8               tty24  tty60    vcs5
gpiochip0               mmcblk0p13          nvhost-vic         ram9               tty25  tty61    vcs6
gpiochip1               mmcblk0p14          nvmap              random             tty26  tty62    vcsa
gpu_freq_max            mmcblk0p2           port               rfkill             tty27  tty63    vcsa1
gpu_freq_min            mmcblk0p3           ppp                rtc0               tty28  tty7     vcsa2
i2c-0                   mmcblk0p4           psaux              rtc1               tty29  tty8     vcsa3
i2c-1                   mmcblk0p5           ptmx               shm                tty3   tty9     vcsa4
i2c-2                   mmcblk0p6           pts                snd                tty30  ttyGS0   vcsa5
i2c-3                   mmcblk0p7           ptyp0              spidev0.0          tty31  ttyS0    vcsa6
i2c-4                   mmcblk0p8           ptyp1              spidev0.1          tty32  ttyS1    vfio
i2c-5                   mmcblk0p9           ptyp2              spidev1.0          tty33  ttyS2    vhci
i2c-6                   mqueue              ptyp3              spidev1.1          tty34  ttyS3    video0
i2c-7                   mtd0                ptyp4              stderr             tty35  ttyTHS1  watchdog
i2c-8                   mtd0ro              ptyp5              stdin              tty36  ttyTHS2  watchdog0
iio:device0             mtdblock0           ptyp6              stdout             tty37  ttyp0    zero
input                   net                 ptyp7              tegra-crypto       tty38  ttyp1    zram0
keychord                network_latency     ptyp8              tegra_camera_ctrl  tty39  ttyp2    zram1
kmem                    network_throughput  ptyp9              tegra_cec          tty4   ttyp3    zram2
kmsg                    null                ptypa              tegra_dc_0         tty40  ttyp4    zram3
loop-control            nvhdcp0             ptypb              tegra_dc_1         tty41  ttyp5
loop0                   nvhost-as-gpu       ptypc              tegra_dc_ctrl      tty42  ttyp6
loop1                   nvhost-ctrl         ptypd              tegra_mipi_cal     tty43  ttyp7
loop2                   nvhost-ctrl-gpu     ptype              tty                tty44  ttyp8
loop3                   nvhost-ctrl-isp     ptypf              tty0               tty45  ttyp9

There's also "video0" :-)

Hoehlenbaer, 2021-05-10:
Solved it!
Instead of a simple "docker run" I used the commands in the enable.sh script and replace the image name with the correct one.
Regards 
Marco