NVIDIA-AI-IOT/jetbot #426: Jetson Nano suddenly turns off during Road Following task

abuelgasimsaadeldin, 2021-04-15:
Hi guys, I am having this problem recently with my Waveshare Jetbot where the Jetson Nano would suddenly turn off mid-way during the road following task. I have already:
1) Changed to a brand new set of batteries
1) Set the Power Mode to 5W
2) Set the fps limit of the camera to 10fps

Is there any other pre-caution steps that I could take to prevent this from occurring again?

Thank you and Best Regards,
Abu

tokk-nv, 2021-04-21:
Hi Abu,
I have not encountered such issues on any Waveshare Jetbot kits, but what you described as actions you already took are in fact what I would do...

Maybe the other thing I would try is to power Jetson Nano Developer Kit through its barrel jack (if you have DC plug 5V-4A power supply and jumper for J48) or through the micro USB port (if you have good microUSB power supply).

abuelgasimsaadeldin, 2021-04-22:
Hi @tokk-nv, thank you for the reply and I agree with you that it highly due to the battery pack from the Waveshare Expansion board is not able to supply a consistent voltage and current required to power on the Jetson Nano and sometimes the Jetson Nano would draw more current than the battery pack can supply so it shuts down.

I would love to test it out with a powerbank and MicroUSB (barrel jack wouldn't be suitable for deployment I believe). My question is to test it out through powerbank which wires should I remove or replace in order to stop supplying power from the battery pack and still have the motor driver connections through I2C? 

[image]

santaimpersonator, 2021-09-30:
I'm not familiar with the waveshare version, but it looks like there is a DC barrel jack next to the opposite antenna. Is that a power input for that board?

cacortespliego, 2022-06-14:
Hello @abuelgasimsaadeldin, I am experiencing the same behavior where the Jetson Nano would suddenly turn off during the road following task. Could you figure out the solution to this?

> Hi guys, I am having this problem recently with my Waveshare Jetbot where the Jetson Nano would suddenly turn off mid-way during the road following task. I have already:
> 
> 1. Changed to a brand new set of batteries
> 2. Set the Power Mode to 5W
> 3. Set the fps limit of the camera to 10fps
> 
> Is there any other pre-caution steps that I could take to prevent this from occurring again?
> 
> Thank you and Best Regards, Abu

davidarodz727, 2023-06-05:
My jetson is having the same exact issue. I have the standard 4GB jetson Nano and I have been experiencing difficulty running code, mainly involving the camera, because it shuts off every time. Any help?