NVIDIA-AI-IOT/jetbot #428: Compatibility with Nano 2GB

tooyipjee, 2021-04-21:
Is there full compatibility with the Nano 2GB? 
At the moment I can't seem to run the example scripts, particularly object detection, using the latest JetBot image. 

The example page claims compatibility with Jetpack 4.3, but the 2GB only has a JetPack 4.4 as it's oldest available JetBot image.

Are there plans to update this discrepancy? 
I would've gotten a 4GB if I knew there were so many version compatibility issues with the 2GB...

jlamondw, 2021-06-24:
I'm seeing the same issue, can't do the object following example with Nano 2GB

santaimpersonator, 2021-07-01:
They haven't built a model that will work with the new Jetpack release. They are working on it. See https://github.com/NVIDIA-AI-IOT/jetbot/issues/430#issuecomment-828633641

santaimpersonator, 2021-09-30:
Just as a note: I was able to get the collision avoidance example to run, but it was **resource intensive**.

* Disabling the graphical desktop freed up enough RAM space to run the example.
* Configuring the power mode to MAX provided more processing power, but it took forever for the code to start running.