NVIDIA-AI-IOT/jetbot #430: Can`t use 'ssd_mobilenet_v2_coco.engine' in jetbot

zoubin2019102976, 2021-04-22:
## My environment: ##
**ubuntu**&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;18.04 aarch64
**jetpack**&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;jetson-nano-jp451-sd-card-image
**jetbot**&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&ensp;&ensp;&ensp;&nbsp;v0.4.3
**tensorflow**&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;2.4.0+nv21.3
**tensorrt**&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;7.1.3.0
**torch**&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&ensp;&ensp;1.6.0
**torchvision**&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;0.7.0a0+78ed10c
**setuptools**&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;49.6.0

## Model version: ##
**ssd_mobilenet_v2_coco.engine:** v0.4 (latest)

## When I execute the command: ##
`from jetbot import ObjectDetector`
`from jetbot import Camera`
`model = ObjectDetector('ssd_mobilenet_v2_coco.engine')`
`camera = Camera.instance(width=300, height=300)`
`detections = model.value(camera.value)`
`print(detections)`

## I get this error information:##
[TensorRT] ERROR: coreReadArchive.cpp (38) - Serialization Error in verifyHeader: 0 (Version tag does not match)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
Traceback (most recent call last):
File "object_follow.py", line 4, in
model = ObjectDetector('ssd_mobilenet_v2_coco.engine')
File "/usr/local/lib/python3.6/dist-packages/jetbot-0.4.3-py3.6.egg/jetbot/object_detection.py", line 29, in init
output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
File "/usr/local/lib/python3.6/dist-packages/jetbot-0.4.3-py3.6.egg/jetbot/tensorrt_model.py", line 59, in init
self.context = self.engine.create_execution_context()
AttributeError: 'NoneType' object has no attribute 'create_execution_context'

## If anyone can help me, I would be very grateful ##

tokk-nv, 2021-04-28:
Hi zoubin2019102976,

Unfortunately, it is an known issue that this model (and the object following example) does not work on the latest JetBot image, as documented in the following page.

http://jetbot.org/v0.4.3/examples/object_following.html
> Please note, the latest object detection engine targets Jetson Nano running software bundled with JetPack 4.3. This engine not be compatible with other versions of JetPack. For this reason, we recommend starting from the JetPack 4.3 JetBot SD Card image, or using the JetPack 4.3 base SD card image if you want to run the object detection example.

You can try with the old JetBot image that is based on the old JetPack 4.3.
`jetbot_image_v0p4p0.zip`
http://jetbot.org/v0.4.3/software_setup/sd_card.html

aixian-jiang, 2021-05-15:
I am having this coco engine issue as well with the jetpck 4.3 (jetbot_image_v0p4p0.zip):
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-78c468b59bd4> in <module>
      1 from jetbot import ObjectDetector
      2 
----> 3 model = ObjectDetector('ssd_mobilenet_v2_coco.engine')

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.0-py3.6.egg/jetbot/object_detection.py in __init__(self, engine_path, preprocess_fn)
     26         load_plugins()
     27         self.trt_model = TRTModel(engine_path, input_names=[TRT_INPUT_NAME],
---> 28                                   output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
     29         self.preprocess_fn = preprocess_fn
     30 

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.0-py3.6.egg/jetbot/tensorrt_model.py in __init__(self, engine_path, input_names, output_names, final_shapes)
     57         with open(engine_path, 'rb') as f:
     58             self.engine = self.runtime.deserialize_cuda_engine(f.read())
---> 59         self.context = self.engine.create_execution_context()
     60 
     61         if input_names is None:

AttributeError: 'NoneType' object has no attribute 'create_execution_context'

anyone can help?

aixian-jiang, 2021-05-15:
issue have been solved by reuploading the the version of ssd_mobilnet_v2_coco.engine

https://github.com/NVIDIA-AI-IOT/jetbot/wiki/examples

jiapei100, 2022-01-20:
@aixian-jiang 

Problem continued in my case:
```console
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-2-78c468b59bd4> in <module>
      1 from jetbot import ObjectDetector
      2 
----> 3 model = ObjectDetector('ssd_mobilenet_v2_coco.engine')

~/.local/lib/python3.6/site-packages/jetbot-0.4.3-py3.6.egg/jetbot/object_detection.py in __init__(self, engine_path, preprocess_fn)
     27         load_plugins()
     28         self.trt_model = TRTModel(engine_path, input_names=[TRT_INPUT_NAME],
---> 29                                   output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
     30         self.preprocess_fn = preprocess_fn
     31 

~/.local/lib/python3.6/site-packages/jetbot-0.4.3-py3.6.egg/jetbot/tensorrt_model.py in __init__(self, engine_path, input_names, output_names, final_shapes)
     57         with open(engine_path, 'rb') as f:
     58             self.engine = self.runtime.deserialize_cuda_engine(f.read())
---> 59         self.context = self.engine.create_execution_context()
     60 
     61         if input_names is None:

AttributeError: 'NoneType' object has no attribute 'create_execution_context'
```

and my jetson is configured as:

```console
➜  ~ jetson_release -v
 - NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack UNKNOWN [L4T 32.5.2]
   * NV Power Mode: MAXN - Type: 0
   * jetson_stats.service: active
 - Board info:
   * Type: Nano (Developer Kit Version)
   * SOC Family: tegra210 - ID:33
   * Module: P3448-0003 - Board: P3542-000
   * Code Name: batuu
   * Boardids: 3448
   * CUDA GPU architecture (ARCH_BIN): 5.3
   * Serial Number: 1423720056255
 - Libraries:
   * CUDA: 10.2.89
   * cuDNN: 8.0.0.180
   * TensorRT: 7.1.3.0
   * Visionworks: 1.6.0.501
   * OpenCV: 4.5.4 compiled CUDA: YES
   * VPI: ii libnvvpi1 1.0.15 arm64 NVIDIA Vision Programming Interface library
   * Vulkan: 1.2.70
 - jetson-stats:
   * Version 3.1.1
   * Works on Python 3.6.9
➜  ~ python --version
Python 3.6.9
```

jiapei100, 2022-01-20:
Things move on. I'm using a **Jetson Nano 2G**.

```console
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-ba8d2e5b664b> in <module>
----> 1 from jetbot import ObjectDetector
      2 
      3 model = ObjectDetector('./ssd_mobilenet_v2_coco.engine')

~/.local/lib/python3.6/site-packages/jetbot-0.4.3-py3.6.egg/jetbot/__init__.py in <module>
----> 1 from .camera import Camera
      2 from .heartbeat import Heartbeat
      3 from .motor import Motor
      4 from .robot import Robot
      5 from .image import bgr8_to_jpeg

~/.local/lib/python3.6/site-packages/jetbot-0.4.3-py3.6.egg/jetbot/camera/__init__.py in <module>
      7     Camera = ZmqCamera
      8 else:
----> 9     from .opencv_gst_camera import OpenCvGstCamera
     10     Camera = OpenCvGstCamera

~/.local/lib/python3.6/site-packages/jetbot-0.4.3-py3.6.egg/jetbot/camera/opencv_gst_camera.py in <module>
      1 import traitlets
      2 import atexit
----> 3 import cv2
      4 import threading
      5 import numpy as np

/usr/local/lib/python3.6/dist-packages/cv2/__init__.py in <module>
    178 
    179 
--> 180 bootstrap()

/usr/local/lib/python3.6/dist-packages/cv2/__init__.py in bootstrap()
    150     py_module = sys.modules.pop("cv2")
    151 
--> 152     native_module = importlib.import_module("cv2")
    153 
    154     sys.modules["cv2"] = py_module

/usr/lib/python3.6/importlib/__init__.py in import_module(name, package)
    124                 break
    125             level += 1
--> 126     return _bootstrap._gcd_import(name[level:], package, level)
    127 
    128 

ImportError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block
```

Looks like there is a **short of memory** issue...

Cheers

closed 2022-07-18 by jaybdub