NVIDIA-AI-IOT/jetbot #431: Printable road for road following

kiwibruyanta, 2021-04-22:
Hello,

i'm looking for a printable road that could be used for the notebook road following. I would like a road similar to the lego tracks used in the videos or similar to the one used in this video https://www.youtube.com/watch?v=ELZWrAmx8jk . Unfortunately i didn't find a model like that on internet.

thank you,

tokk-nv, 2021-04-28:
Hello kiwibruyanta,
Since you are collecting the data to train your model by yourself as well, you don't need to stick with the LEGO plate, but you can use any design road plate you like.
So you can draw couple of patterns using any drawing software you like (GIMP, Inkscape being free options), and print them out.

An article like the below may give you more ideas.
https://www.whilehewasnapping.com/printable-diy-race-track-kids-love

DingoOz, 2021-04-29:
I used black building plastic and masking tape to create a road that can pack away easily: https://adventuresinsilicon.blogspot.com/2021/01/cheap-and-easy-robot-racetrack-or-line.html

AnbuKumar-maker, 2021-05-02:
> Hello,
>
> i'm looking for a printable road that could be used for the notebook road following. I would like a road similar to the lego tracks used in the videos or similar to the one used in this [...]

Hello, you may use the attached PDF to print a nice race track road, I used this to train my JetBot for Road Following
[https://github.com/AnbuKumar-maker/Printable-Road-For-NVIDIA-JetBot](url)

closed 2022-07-18 by jaybdub