NVIDIA-AI-IOT/jetbot #436: Jetbot Object Following

qlawyer, 2021-05-01:
Hello, 

I want to run Jetbot Object Following Example. I used jetbot v0.4.0, as jetpack 4.3 version is recommended for it. I follow the SD card flash steps, copied the v0.3 | ssd_mobilenet_v2_coco.engine to object following folder and run the first cell, got the following error: 

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-78c468b59bd4> in <module>
      1 from jetbot import ObjectDetector
      2 
----> 3 model = ObjectDetector('ssd_mobilenet_v2_coco.engine')

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.0-py3.6.egg/jetbot/object_detection.py in __init__(self, engine_path, preprocess_fn)
     26         load_plugins()
     27         self.trt_model = TRTModel(engine_path, input_names=[TRT_INPUT_NAME],
---> 28                                   output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
     29         self.preprocess_fn = preprocess_fn
     30 

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.0-py3.6.egg/jetbot/tensorrt_model.py in __init__(self, engine_path, input_names, output_names, final_shapes)
     57         with open(engine_path, 'rb') as f:
     58             self.engine = self.runtime.deserialize_cuda_engine(f.read())
---> 59         self.context = self.engine.create_execution_context()
     60 
     61         if input_names is None:

AttributeError: 'NoneType' object has no attribute 'create_execution_context' 

Same happened in the latest version as well. Do I miss something? Thank you for your help in advance.

gwiheo, 2021-05-08:
https://forums.developer.nvidia.com/t/jetson-nano-jetbot-ssd-mobilenet-v2-coco-engine-has-no-attribute/107397
Pls refer the above link. If you use jetpack 4.3 version with correct version of ssd_mobilenet_v2_coco.engine, the problem will be solved.

qlawyer, 2021-05-11:
Thanks for the help. I changed my design because jetbot comes with docker even if I use the SD card image and it makes the build procedure uncomfortable -for my experience level-.

closed 2021-05-11 by qlawyer