NVIDIA-AI-IOT/jetbot #440: Jetbot waveshare

futurenowx, 2021-05-07:
Hi,

i bought a jetbot from waveshare and found a couple of problems. 

**1. jetpack problems**

I downloaded all jetpack versions 
https://jetbot.org/v0.4.3/software_setup/sd_card.html

1. from jetpack 4.4.1 no ubuntu desktop
2. jetpack 4.2 do not start
3. jetpack 4.3 immensive slow

the only jetpack which is working is the one you can download from waveshare

2. problems with road following

i followed the nvidia video Jetson AI Fundamentals - S2E4 - JetBot Road Following
https://www.youtube.com/watch?v=VY7m4lPIOj0&list=RDCMUCBHcMCGaiJhv-ESTcWGJPcw&index=8
i use the lego road 
i took 200 images exactly as document and created a model with 70 epochs also one with 100
but still the robot is not follwing the line.

All other projects are mainly working

may anybody has experience with same issues and let me know how you solved it.

thx

Alexander

gwiheo, 2021-05-08:
Recently I tested my Waveshare jetbot using Jetpack 4.5.1 and 4.3. It worked good for obstacle avoidance and road following. I found that object following did not work due to the slow speed of Jetson nano (I think). I tested road following with Track Map which included at my PiRacer from WaveShare. (https://www.waveshare.com/product/ai/robots/mobile-robots/piracer-ai-kit.htm) For road following you need to optimize speed gain and steer gain (at the beginning set it very low as 0.01, then slowly increase the gain)

futurenowx, 2021-05-09:
> Recently I tested my Waveshare jetbot using Jetpack 4.5.1 and 4.3. It worked good for obstacle avoidance and road following. I found that object following did not work due to the slow speed of [...]

Hi 

thx for your support. I tried 4.3 but its quite slow and 4.5.1 has no ubuntu desktop, may you faced same problem, which would be no problem as long you want to work only from command line.  i played already with the different speed and steer gains but will invest some more time for trial.
Seems the nvidia jetpacks are not compatible with all tutorials, which is quite bad.

thx 

Alexander

aixian-jiang, 2021-05-15:
Hi futurenowx,

Have you solved the problem? I am facing the same issue -- speed too slow! can share your solution if you settled the problem? Thanks! 

Best Regards,
aixian

futurenowx, 2021-05-17:
> Hi futurenowx,
> 
> Have you solved the problem? I am facing the same issue -- speed too slow! can share your solution if you settled the problem? Thanks!
> 
> Best Regards,
> aixian

sorry for late reply, no i did not test it till now, i was busy with a different project.  will may do it tonight.
what kind of jetbot you exactly use, which jetpack and notebooks from where.

abuelgasimsaadeldin, 2021-05-18:
Hi guys, may I know instead of trying out different JetPack versions why don't you just use the Jetbot image that is provided by Waveshare for their Jetbot here:
https://drive.google.com/file/d/1G5nw0o3Q6E08xZM99ZfzQAe7-qAXxzHN/view