NVIDIA-AI-IOT/jetbot #446: Mysterious message at initial startup |  jetbot-043_nano-4gb

Taitan2010, 2021-05-20:
For my Jetson Nano (4GB) Dev Kit - Revision B01,  I have been using jetbot-043_nano-4gb-jp45.zip as noted on this page.
http://jetbot.org/v0.4.3/software_setup/sd_card.html#step-1-download-the-pre-built-jetbot-sd-card-image

When starting power supply, CUI will be displayed instead of the expected GUI, so I enter Login name; jetbot and password: jetbot. Then enter the 'unminimize' command and 'y' according to the message suggestion, but eventually the 'jetson nano tegra-i2c 7000c400.i2c: no accnowledge from address' message is repeated.

How can I set it up with GUI? This is not a hardware problem, since I have checked by using 3 machine and 2 SD cards.

[image]

dgr1992, 2021-05-21:
I am experiencing the exact same thing after switching to the latest Jetbot image.

Taitan2010, 2021-05-23:
Dear Daniel 

Thank you for your advice. I will appreciate if you could let me know how to solve this issue. I tried to check another version as well but it was same result unfortunately.
Best Regards 

Tatsuyuki Morikawa 

iPhoneから送信

dgr1992, 2021-05-23:
I don't have a solution yet but I think the problem is the same as mentioned in #385 and #390. So you should close this topic and refer to those two issues. At least that's what I'm going to do.

Taitan2010, 2021-05-23:
I got it. Thank you for your comment. 
Best Regards 

iPhoneから送信

dgr1992, 2021-05-25:
So first thing is that by default the UI is disabled, I missed that information my self. The other thing is the i2c error, it origins from the jetbot/apps/stats.py which tries to connect to the OLED display in a loop every 10sec. To get rid of the message you can modify the section like this

```
while True:
    try:
        # Try to connect to the OLED display module via I2C.
        disp.begin()
    except OSError as err:
        print("OS error: {0}".format(err))
        exit()
    else:
        break
```

h3retek, 2021-07-02:
> I just realiesed that above Code has no effect as everything is running in a docker.
> There is a docker called "jetbot_display" and it seems like it is responsible for showing information on the OLED display.
> You can try following command to deactivate this docker if you don't need the OLED display `sudo docker update --restart=no jetbot_display`
> 
> Also I would recommend to set the topic to **Closed** if this solved your problem.

Well, this definitely solved it.

Thank you... how did you figure it out?

dgr1992, 2021-07-02:
I figured it out by doing some research about the jetbot image where I came across that the Jupyter Notebook is running in a docker and a second docker hosts all other hardware related things.

ThaiVuz, 2022-06-02:
> For my Jetson Nano (4GB) Dev Kit - Revision B01, I have been using jetbot-043_nano-4gb-jp45.zip as noted on this page. [...]

I am facing the same situation. I can't log on to display mode. CUI will be displayed instead of the expected GUI. Do you know how to fix it? Please help me !!

closed 2022-07-15 by jaybdub