NVIDIA-AI-IOT/jetbot #447: Can't initialize LXDE

axteroide, 2021-05-21:
Hi.
I have the 2GB Jetson with the Waveshare jetbot.
I installed in my SD card the last image that cames without desktop. I've been trying to install and use LXDE without any success.
For now I've tried this:
 - https://github.com/jetsonhacks/installLXDE
 - the unminimize command
 - dpkg-reconfigure lightdm

Any of those seems to do nothing at all, the system keeps booting on command line mode.
Someone has been able to initialize it on desktop mode?

Thank you.

Hoehlenbaer, 2021-05-24:
Hi axteroide,

Why do you want a GUI on the jetbot? Actually this will bring you to the base system, but with the newer versions, the magic is within the docker containers.

I spent a lot of time to have a working RDP, but now I only use the jupyterlab interface.
Regards
Marco

axteroide, 2021-05-25:
I intend to use ROS to test some things, some of them, like ORB-SLAM requires a working GUI because is using OpenCV windows system, so it won't work with only jupyter.

santaimpersonator, 2021-06-17:
They probably disabled the GUI with the [script here](https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/scripts/configure_jetson.sh#L4), try the following commands:

> `sudo systemctl set-default graphical.target`
> `sudo reboot` *(needed for changes to take effect)*

Otherwise, you can also run this [script](https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/scripts/re_enable_gui.sh) to enable the GUI.

closed 2022-07-18 by jaybdub