NVIDIA-AI-IOT/jetbot #461: Only 24G storage is used in 64G -SD card

atyouspace, 2021-06-22:
I downloaded "jetbot-043_nano-4gb-jp45.zip" from [v0.4.3](https://github.com/NVIDIA-AI-IOT/jetbot/releases/tag/v0.4.3) for jetbot (Jetson Nano (4GB)), and flash it by etcher successfully. But when I power jetbot, only 24G storage is available. 
What should I do ？Is there another image with different storages?
[image]

santaimpersonator, 2021-06-22:
You can run this [script](https://github.com/NVIDIA-AI-IOT/jetcard/blob/master/scripts/archive/nvresizefs.sh) from the jetcard repository to resize the partition.

chaiai, 2021-06-23:
I found an easy way to fix this is by opening up GParted right after the image is flashed, but before ejecting the SD card from the host machine. I'm unsure about other programs, but when I opened GParted, it gave me the warning to fix the unused space on the SD card. Click yes and then simply right-click on the APP partition and resize it to the maximum size, then click the green checkmark to start it.

TCIII, 2021-06-23:
Try lsblk:

jetbot@nano-4gb-jp45:~$ lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
loop0          7:0    0   16M  1 loop
mtdblock0     31:0    0    4M  0 disk
mmcblk0      179:0    0 59.5G  0 disk
├─mmcblk0p1  179:1    0 24.4G  0 part /
├─mmcblk0p2  179:2    0  128K  0 part
├─mmcblk0p3  179:3    0  448K  0 part
├─mmcblk0p4  179:4    0  576K  0 part
├─mmcblk0p5  179:5    0   64K  0 part
├─mmcblk0p6  179:6    0  192K  0 part
├─mmcblk0p7  179:7    0  384K  0 part
├─mmcblk0p8  179:8    0   64K  0 part
├─mmcblk0p9  179:9    0  448K  0 part
├─mmcblk0p10 179:10   0  448K  0 part
├─mmcblk0p11 179:11   0  768K  0 part
├─mmcblk0p12 179:12   0   64K  0 part
├─mmcblk0p13 179:13   0  192K  0 part
└─mmcblk0p14 179:14   0  128K  0 part

tokk-nv, 2021-07-29:
Hi,
This is due to an issue with JetCard partition resize script, which is fixed in the latest version.
Would you manually execute the following to update the local `jetcard` and trigger the resize?

```
cd ~/jetcard
git pull
./script/jetson_install_nvresizefs_service.sh
```

TCIII, 2021-07-30:
@tokk-nv,

Unfortunately you have an error in your script command: 
S/B: ./script**s**/jetson_install_nvresizefs_service.sh

Once I corrected the error, your script ran, but nothing changed according to df -h:

thomas@thomas-desktop:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1  118G   23G   91G  20% /
none            1.8G     0  1.8G   0% /dev
tmpfs           2.0G   52K  2.0G   1% /dev/shm
tmpfs           2.0G   20M  2.0G   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           2.0G     0  2.0G   0% /sys/fs/cgroup
tmpfs           397M     0  397M   0% /run/user/1000

The above is the same as before running your script.

Comments?

TCIII, 2021-07-30:
@tokk-nv,

Unfortunately you have an error in your script command:
S/B: ./script**s**/jetson_install_nvresizefs_service.sh

Also, you need to reboot the Nano after running the script for the resize to take place.

Once I corrected the error, your script ran, and here is the result of running df -h: 
 
jetbot@nano-4gb-jp45:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1   59G   22G   35G  39% /
none            1.8G     0  1.8G   0% /dev
tmpfs           2.0G  4.0K  2.0G   1% /dev/shm
tmpfs           2.0G   19M  2.0G   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           2.0G     0  2.0G   0% /sys/fs/cgroup
tmpfs           397M     0  397M   0% /run/user/1000

As you can see there is now the full 59GB showing with 35GB in use compared to atyouspace's df output.

Issue resolved.