NVIDIA-AI-IOT/jetbot #466: Determine storage location of Adafruit_MotorHAT library package in Jetbot v 4.3

TCIII, 2021-06-25:
Need a little assistance here:

The Jetbot v 4.3 image communicates with the FeatherWing Motor Controller board using the Adafruit_MotorHAT library. Prior to Jetbot v 4.3 the Adafruit_MotorHAT library package was stored at “/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py” The Jetbot v 4.3 image no longer stores the Adafruit_MotorHAT library package at that location as “/usr/local/lib/python3.6/dist-packages/" is empty in the Jebot v 4.3 image.
Therefore I would like to know where the Jetbot v 4.3 image is storing the Adafruit_MotorHAT library package sol that I can adjust the PCA9685 servo board’s PWM switching frequency as I have stated above.

Reply
cancel

TCIII, 2021-06-25:
I have received the following from Adafruit Customer Service:

That library is deprecated and has been archived:
https://github.com/adafruit/Adafruit-Motor-HAT-Python-Library

The new library is here:
https://github.com/adafruit/Adafruit_CircuitPython_MotorKit

Install info in guide here:
https://learn.adafruit.com/adafruit-dc-and-stepper-motor-hat-for-raspberry-pi/installing-software

Comments?

santaimpersonator, 2021-06-28:
As of last week, I didn't have issues installing the jetbot package (includes installation of Adafruit Motor Hat package) with the [setup.py script](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Software-Setup#step-5---install-latest-software-optional) or with the [Docker containers](https://github.com/NVIDIA-AI-IOT/jetbot/tree/master/docker).

TCIII, 2021-06-28:
@santaimpersonator,

I never had a problem either installing Jetbot from the latest Nano 4GB image, but that is not the subject of this issue.

You will notice that Adrafruit_MotorHAT is deprecated and therefore needs to be replaced with the Adafruit_CircuitPython_MotorKit library as advised above.

However, I digress. All I want to know is the following:

"The Jetbot v 0.4.3 image communicates with the FeatherWing Motor Controller board using the Adafruit_MotorHAT library. Prior to Jetbot v 0.4.3 the Adafruit_MotorHAT library package was stored at “/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py” The Jetbot v 0.4.3 image no longer stores the Adafruit_MotorHAT library package at that location as “/usr/local/lib/python3.6/dist-packages/" is empty in the Jebot v 0.4.3 image.
Therefore I would like to know where the Jetbot v 0.4.3 image is storing the Adafruit_MotorHAT library package sol that I can adjust the PCA9685 servo board’s PWM switching frequency."

santaimpersonator, 2021-06-29:
To clarify:
- The installation steps and Python package work fine. Therefore, it isn't necessary to utilize the other package.
    - It sounds like *maybe* want to file an issue about utilizing the new Python package (that isn't deprecated) instead
- Additionally, I was showing you how the packages got installed (to point you in the right direction).

That being said...
- It sounds like previously, you used this repo as a Python package (via the [setup.py script](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Software-Setup#step-5---install-latest-software-optional)), which installs the Adafruit package with the rest of the Python packages.
    - If you aren't familiar with Python, the files may also be in the hidden local folder of the home directory.
    - I'm fairly confident that when packages are installed the printout tells you where they area located, as well.
- The recent Jetbot implementations use Docker containers to create their own "virtual workspace".
    - You can modify the files from within the Docker container, but any changes you make will not be permanent.
        - The location of the "virtual workspace" Python files will be similar to OS
    - For reference, the [setup.py script](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Software-Setup#step-5---install-latest-software-optional) still installs the Adafruit Python packages for the docker container: https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/setup.py#L20

*Good luck making all the necessary changes to utilize the new Adafruit package. You'll want to make all you code changes before [building and enabling your container](https://github.com/NVIDIA-AI-IOT/jetbot/tree/master/docker).*

TCIII, 2021-06-29:
@santaimpersonator,

You have totally missed the point. 

If you install the Jetbot IMAGE from [here,](https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/docs/software_setup/sd_card.md) there are no "packages" to watch being built and installed as the required Jetbot packages/modules are already in the IMAGE. 
You need to understand the difference between a prebuilt Jetbot IMAGE and an installation of Jetbot built from SCRATCH by the user which I did not have to do in this case.
When I first used Jetbot, I had to first install Jetpack and then follow a guide to install the required Jetbot packages and modules.
Example: I have built Jetrace from scratch using these [instructions](https://github.com/NVIDIA-AI-IOT/jetracer/issues/89) and as a completed image found [here](https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/software_setup.md) so I know all about installations from scratch and from images.

TCIII, 2021-06-29:
@santaimpersonator,

You didn't bother to read my response at all, probably stopping at "my response was pointless" as you obviously do not know the difference between an installed image of Jetbot and an Jetbot installation built from scratch.

Docker has nothing to do with my issue and therefore you have nothing to bring to the table to address my issue.

TCIII, 2021-07-06:
@santaimpersonator,

Okay, I was able to get to where "Adafruit_MotorHAT_Motors.py" is in the jetbot workspace using the Jupyter terminal.

However, how can I edit "Adafruit_MotorHAT_Motors.py" as gedit only works in the Desktop mode and there is no nano editor in the workspace?

closed 2021-07-06 by TCIII