NVIDIA-AI-IOT/jetbot #473: Jetbot issue

calleliljedahl, 2021-08-19:
Hey guys. I got my jetbot ("_SparkFun JetBot AI Kit v2.1 Powered by Jetson Nano_") about a month ago and I just cant get it to work. Im really struggling to know what to do next - it feels like I've tried all i could... 

I just redid the hardware assembly for the third time - no problems at all there. Now for the flashing of the SD card: I have tried three different images: 
- "jetbot_image_v0p4p0.zip"
- "jetbot_image_v0p3p2.zip"
- "jetson-nano-jp46-sd-card-image.zip"
- "jetson-nano-sd-card-image-r32.2.zip"

currently im running on the "jetbot_image_v0p4p0.zip" for the second time. 
After assembly is done I create a docker file and update the system - allowing me to use the edumax-wifi dongle. I can run the jetson both through the GUI using HDMI, aswell as through ssh. 

What I cannot do though, is to initialize the robot class - or run anything that moves the robot really. Also the LED display does not turn on. 

Running Ubuntu 18.04.5 LTS

The error message i get whenn trying to run 
`from jetbot import Robot
robot = Robot()`

is the following:
`---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-2-b418ad9f6ab3> in <module>
----> 1 robot = Robot()

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.3-py3.6.egg/jetbot/robot.py in __init__(self, *args, **kwargs)
     20     def __init__(self, *args, **kwargs):
     21         super(Robot, self).__init__(*args, **kwargs)
---> 22         self.motor_driver = Adafruit_MotorHAT(i2c_bus=self.i2c_bus)
     23         self.left_motor = Motor(self.motor_driver, channel=self.left_motor_channel, alpha=self.left_motor_alpha)
     24         self.right_motor = Motor(self.motor_driver, channel=self.right_motor_channel, alpha=self.right_motor_alpha)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py in __init__(self, addr, freq, i2c, i2c_bus)
    229         self.motors = [ Adafruit_DCMotor(self, m) for m in range(4) ]
    230         self.steppers = [ Adafruit_StepperMotor(self, 1), Adafruit_StepperMotor(self, 2) ]
--> 231         self._pwm = PWM(addr, debug=False, i2c=i2c, i2c_bus=i2c_bus)
    232         self._pwm.setPWMFreq(self._frequency)
    233 

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in __init__(self, address, debug, i2c, i2c_bus)
     57         self.i2c = get_i2c_device(address, i2c, i2c_bus)
     58         logger.debug("Reseting PCA9685 MODE1 (without SLEEP) and MODE2")
---> 59         self.setAllPWM(0, 0)
     60         self.i2c.write8(self.__MODE2, self.__OUTDRV)
     61         self.i2c.write8(self.__MODE1, self.__ALLCALL)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in setAllPWM(self, on, off)
     93     def setAllPWM(self, on, off):
     94         "Sets a all PWM channels"
---> 95         self.i2c.write8(self.__ALL_LED_ON_L, on & 0xFF)
     96         self.i2c.write8(self.__ALL_LED_ON_H, on >> 8)
     97         self.i2c.write8(self.__ALL_LED_OFF_L, off & 0xFF)

/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.4-py3.6.egg/Adafruit_GPIO/I2C.py in write8(self, register, value)
    114         """Write an 8-bit value to the specified register."""
    115         value = value & 0xFF
--> 116         self._bus.write_byte_data(self._address, register, value)
    117         self._logger.debug("Wrote 0x%02X to register 0x%02X",
    118                      value, register)

/usr/local/lib/python3.6/dist-packages/Adafruit_PureIO-1.0.4-py3.6.egg/Adafruit_PureIO/smbus.py in write_byte_data(self, addr, cmd, val)
    266         # Send the data to the device.
    267         self._select_device(addr)
--> 268         self._device.write(data)
    269 
    270     def write_word_data(self, addr, cmd, val):

OSError: [Errno 121] Remote I/O error`

Thanks in advance me and my son are desperate at this point. What do you think I could try next?

calleliljedahl, 2021-08-20:
Does this help at all? This guy says they should be on 60 and 70... Whats going on here? https://github.com/NVIDIA-AI-IOT/jetbot/issues/45#issuecomment-537583380
When i run `sudo i2cdetect -y -r 1` i get: `
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- 3d -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- 5d -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   `

santaimpersonator, 2021-09-05:
The problem is that SparkFun modifies the Jetbot code to support the hardware in our kit. That currently, isn't supported with this repository and the images provided by Nvidia yet; we have a pull request pending: https://github.com/NVIDIA-AI-IOT/jetbot/pull/467

---

**Note:** You might be able to use the images from [our assembly guide](https://learn.sparkfun.com/tutorials/assembly-guide-for-sparkfun-jetbot-ai-kit-v20/4-software-setup-guide-from-nvidia) *(see the troubleshooting note, for users who overwrote our custom image from their SD card)*. However, I can't remember off the top of my head if those images work with the new firmware that you loaded on your board *(see the **Attention** note from the [jetbot.org website](http://jetbot.org/master/software_setup/sd_card.html) about the "one-time configuration")*.

---

Unfortunately, you used a Jetpack 4.5+ image, which means you updated the firmware on the board and only certain images will work with the board now. That also complicates things to get the board working. I am currently, working on a supplementary guide for more details on the instructions below:

Until our pull request is merged, you will need to use our repository:  https://github.com/sparkfun/jetbot/
* Since you have updated the firmware, I recommend using this image for the 4GB board from the [jetbot.org website](http://jetbot.org/master/software_setup/sd_card.html):
    * [jetbot-043_nano-4gb-jp45.zip](https://drive.google.com/file/d/1o08RPDRZuDloP_o76tCoSngvq1CVuCDh/view?usp=sharing)
* You will need to adopt the changes from these branches:
    * https://github.com/sparkfun/jetbot/tree/update-SparkFun_hardware - Adds the code to support our hardware
        * You will also need to add the commit for the `utils.py` file for the changes to work properly: https://github.com/NVIDIA-AI-IOT/jetbot/commit/436755b244f3a8736b5b73417322f99b0a420bd9
    * https://github.com/sparkfun/jetbot/tree/update-Docker - Addresses the docker image issue reported here: https://github.com/NVIDIA-AI-IOT/jetbot/issues/406
        * You will need to rebuild the Docker container (process takes a few hours)

It's a little messy with the two branches, but I did it that way to break up the pull requests properly. Essentially, you need to:
1. Use the image from above: [jetbot-043_nano-4gb-jp45.zip](https://drive.google.com/file/d/1o08RPDRZuDloP_o76tCoSngvq1CVuCDh/view?usp=sharing)
2. Disable and remove the running docker containers
    * This is **super critical**!
3. Uninstall the jetbot Python package
4. Copy the `utils.py` file into a separate location, then delete the `jetbot` folder
5. Pull/Clone the [branch with the Docker container fixes](https://github.com/sparkfun/jetbot/tree/update-Docker)
6. Copy the `docker` folder into a separate location, then delete the `jetbot` folder
7. Pull/Clone the [branch with our hardware support](https://github.com/sparkfun/jetbot/tree/update-SparkFun_hardware)
8. Replace the `docker` folder and `utils.py` file, with the one you set aside
9. Don't forget to download the pre-trained models for the examples (as mentioned here: https://github.com/NVIDIA-AI-IOT/jetbot/wiki/examples)
10. Rebuild the docker containers
    * This takes a few hours, so leave the Jetson Nano plugged in with a power supply. The battery will not last!
    * Also, configure the board for max power, it will complete about twice as fast than if it were in `5W mode`

Hopefully, I didn't miss anything... but this should be enough to get you started. This doesn't include the driver support for the wifi/bluetooth adapter in our kit. You will need to follow this guide: https://learn.sparkfun.com/tutorials/adding-wifi-to-the-nvidia-jetson#driver-installation

2500035435, 2021-10-09:
Thank you so much@santaimpersonator，but I just noticed I can't boot image jetbot-043_nano-4gb-jp45.zip, and meet the same problem as https://github.com/NVIDIA-AI-IOT/jetbot/issues/343
It means I can't enter the GUI desktop and have an error: tegra-i2c 7000c400.i2c: no acknowledge from address 0x3c
I am sure the OLED pin have been inserted into the board

So I am going to back to https://pan.baidu.com/s/1Kq3YpQeBLY6W1UTsM-yZuA?from=init  (code: t64u)
AND still meet the "Could not initialize camera. Please see error trace."  and "robot = Robot() [Errno 121] Remote I/O error"
It was a desperate day : (

santaimpersonator, 2021-10-11:
@2500035435 Maybe it is a translation issue, but I am not following where you are going with your issue. Your current problem seems to be completely unrelated to your original post.

The original issue for this thread was related, specifically to the [SparkFun Jetbot Kit](https://www.sparkfun.com/products/18486)... which is where [my specific instructions helps](https://github.com/NVIDIA-AI-IOT/jetbot/issues/473#issuecomment-913087787).

For the Nvidia version of the kit, I recommend following the instructions on [jetbot.org](http://jetbot.org/master/) and using the [Nvidia forum](https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-nano/76) for technical support.

RyanRReese, 2021-12-30:
@santaimpersonator Having the same issue, going to follow your instructions above. Kind of frustrating that instructions sparkfun provides tell you to update the jetbot version using NVIDIA's github which breaks the code... https://learn.sparkfun.com/tutorials/assembly-guide-for-sparkfun-jetbot-ai-kit-v20/4-software-setup-guide-from-nvidia

santaimpersonator, 2021-12-31:
@RyanRReese Could you file a new issue, in [our fork of this repository](https://github.com/sparkfun/jetbot)... just because this issue already has multiple topics and it would help with tracking your particular problem. Please include exact details on the problem you are having and some debug information for me (steps you performed so far, bash history, jetbot image version, if you have the docker implementation, etc.). Thanks!

I see the problem in the "update" instructions. Sorry about that, I'll reword that bullet point. Unfortunately, that tutorial has to cover two different implementations and sets of instructions.

* I have been working on a separate hookup guide for the software instructions to address the changes of the software implementation. Unfortunately, there's a lot of information to cover and compact... so it is still in progress.

closed 2022-07-18 by jaybdub