NVIDIA-AI-IOT/jetbot #475: Finding a suitable driver and a shield for jetson nano.

ntt160402, 2021-08-29:
I have a project which required using jetson nano to control 4 stepper motors for a robot car. I have been looking for a suitable driver to connect between the Jetson and the motor for weeks but I'm totally lost. I literally have no clues about this project, so I need some help. If you have done this project before or you have an idea of it please teach me. I appreciate all.

Also I have heard about something called the CNC shield (for Arduino). I wonder if those kind of shield can be connected to Jetson? If you have any experiences, please let me know. I would be really really grateful for any help.

santaimpersonator, 2021-09-05:
### This probably belongs in the [Discussions tab](https://github.com/NVIDIA-AI-IOT/jetbot/discussions) or the [Nvidia forum](https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-nano/76).

You will need to code or use a Python package that supports your hardware, then you will need to adapt that code into these files of the jetbot code:
* https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/jetbot/motor.py
* https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/jetbot/robot.py

***Note:** If you are using the version with the Docker container, you will need to disable/remove the containers and the re-build the container to adopt the changes.*

It sounds like you might be in over you head on this. However, if you decide to sink a ton of time into pursuing it... the [Nvidia forum](https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-nano/76) might be your best option for help. These tips are the most I can offer:
* I would recommend using a [release prior to the Docker implementation](https://github.com/NVIDIA-AI-IOT/jetbot/releases/tag/v0.4.0), unless you are familiar with Docker containers
* I would also recommend finding a Python package that supports using stepper motors and adapt that code into the files above. That way only Python is being used.
    * The [Adafruit motor driver](https://www.adafruit.com/product/2927), which is the recommended hardware, supports the operation of stepper motors (from product page pic)
        * Here is their tutorial for that product: https://learn.adafruit.com/adafruit-stepper-dc-motor-featherwing
        * I'm not sure if it is limited to 2 stepper motors without digging into it
        * You may probably need to upgrade your battery pack. Stepper motors tend to consume more power than the recommended hobby motors for this kit.
* I recommend dev testing with iPython3... you can make code changes in real-time without having to re-run files.

ntt160402, 2021-09-05:
> ### This probably belongs in the [Discussions tab](https://github.com/NVIDIA-AI-IOT/jetbot/discussions) or the [Nvidia [...]

Thank you so much. I really really appreciate all the opinions you gave me. Have a good day sir.

closed 2022-07-18 by jaybdub