NVIDIA-AI-IOT/jetbot #479: The output on the powerbank, which is responsible for the motor driver, is switched off after a short time.

91fabbai91, 2021-09-17:
Hey guys,

the output on the powerbank, which is responsible for the motor driver, is switched off after a short time. It seams that it's the result of low current used by the motor shield. In the EU, it is not possible to get power banks without this "feature. So does anyone have a solution to this? Maybe to send some current to the motors or something simillar?

Or does anyone have a recommendation for a powerbank, which has the size to fit into the jetbot and also has a permanently powered port?

Usimian, 2021-09-20:
It seems most of the USB batteries you find on Amazon use the same chipset.  I have two INIU powerbanks and only one will stay on with the low power draw of the motor shield.  The other big problem with these powerbanks is they don't work as a battery backup for Arduino/Jetson stuff.  If you have the powerbank charging and the wall power goes out, the powerbank will momentarily shutoff which depowers the computer.

My solution was a LiPo with a Buck converter like https://www.amazon.com/gp/product/B085T73CSD/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

santaimpersonator, 2021-09-30:
Is there a way to provide external power to the motor driver?
* There are 5V and 3.3V pins on the Jetson. They don't source a ton of current, but you probably can pull ~200-500mA without browning out the board.
* Otherwise, we offer this power bank, but in needs at least 20mA of current draw *(a lot lower than the minimum current draw of other power banks)* or it'll shut off after a period of time:
    * https://www.sparkfun.com/products/15204