NVIDIA-AI-IOT/jetbot #486: imeoutError: [Errno 110] Connection timed out" error using Adafruit Jetbot

Taitan2010, 2021-10-16:
**What I want to solve:**

I want to initialize a Robot class instance with Jetbot.

I'm making a Jetbot with NVIDIA's Jetson Nano. I created the hardware and installed the software according to the GitHub manual. I am getting the following error while handling a Basic Motion program with Collision Avoidance in Jupyter Notebook. Please let me know the solution.

**Problems / errors that are occurring**
[image]

**What I have already checked**
1. Jetbot appli.: No problem. Collision Avoidance Train Model was completed.
2. Motion of the motor itself. : Work when connecting battery directly.
3. Power supply to motor driver: Always Green lights
4. Camera (other kernel): No problem. Data Collection of Collision Avoidance is completed
5. The PiOLED display: Did not show anything. So I removed this display and connected the Jetson Nano with  the Motor Driver directly.

santaimpersonator, 2021-11-23:
You probably have a wiring issue with the I2C connection, since the OLED and motor driver don't show up; make sure you don't have your SDA and SCL lines mixed up. Also, the camera doesn't show up on that I2C bus.

Usimian, 2021-11-27:
You can try hooking up the pioled to the other I2c port.  I have a Nano that has a bad I2c bus 0 but everything else including bus 1 works fine.  They’re sensitive to mis-wiring.

You might have a permission issue also.