NVIDIA-AI-IOT/jetbot #495: cannot use object detection notebook

hessio, 2021-11-10:
When I try to run the object detection in the live_demo.ipynb I get the below error

AttributeErrorTraceback (most recent call last)
<ipython-input-12-78c468b59bd4> in <module>
      1 from jetbot import ObjectDetector
      2 
----> 3 model = ObjectDetector('ssd_mobilenet_v2_coco.engine')

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.3-py3.6.egg/jetbot/object_detection.py in __init__(self, engine_path, preprocess_fn)
     27         load_plugins()
     28         self.trt_model = TRTModel(engine_path, input_names=[TRT_INPUT_NAME],
---> 29                                   output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
     30         self.preprocess_fn = preprocess_fn
     31 

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.3-py3.6.egg/jetbot/tensorrt_model.py in __init__(self, engine_path, input_names, output_names, final_shapes)
     57         with open(engine_path, 'rb') as f:
     58             self.engine = self.runtime.deserialize_cuda_engine(f.read())
---> 59         self.context = self.engine.create_execution_context()
     60 
     61         if input_names is None:

AttributeError: 'NoneType' object has no attribute 'create_execution_context'

santaimpersonator, 2021-11-23:
- Did you copy over the `ssd_mobilenet_v2_coco.engine` pre-trained model into that folder?
  [image]
    - If not you can download it from here (click on the _pre-trained model_ hyper link): http://jetbot.org/master/examples/collision_avoidance.html
- Which image are you using?
    - Are you using the image from here: http://jetbot.org/master/software_setup/sd_card.html

petex8989, 2021-11-25:
I am also having this issue. I am using the latest image of jetbot (0.4.3), and it is not working. I have tried both versions of ssd_mobilenet_v2_coco.engine, and both give the same error.

jmadrover, 2021-11-26:
Same here, tried all engine files I could find with no luck... any clue?

santaimpersonator, 2021-11-29:
Sorry, I just realized that you guys are trying to use the **Object Following** example and not the **Collision Avoidance** example... *you really should copy the example name directly to avoid confusion*

They haven't built a new engine for JetPack 4.5. This is a duplicate of issue https://github.com/NVIDIA-AI-IOT/jetbot/issues/430. Please close this issue.

Tiryoh, 2021-12-28:
I guess maybe you can try using `build_engine_from_tensorflow.ipynb` to build the engine file for TensorRT 7 (JetPack 4.5).

https://github.com/rt-net/jnm_jupyternotebook/blob/master/notebooks/object_following/build_engine_from_tensorflow.ipynb