NVIDIA-AI-IOT/jetbot #497: Teleoperation issue: Firefox on linux, Gamepad recognized on gamepad test website, but notebook says gamepad not supported in browser

thesaltymaker, 2021-11-15:
Firefox on linux, Gamepad recognized on gamepad test website, but notebook says gamepad not supported in browser

The gamepad is supported and found on https://gamepad-tester.com/, but the jupyter notebook for teleoperation says gamepads are not supported in this browser.  Any way around this issue.  We don;t have windows computers in our home.  

Unfortunately chromium does not detect the gamepad at all. 

thanks for any help.

santaimpersonator, 2021-11-23:
Can you share a screen shot of the https://gamepad-tester.com/ website showing the gamepad is supported?
Can you share where the code fails in the notebook (code and error message)?

* Make sure you changed the parameter `index=1` in the notebook to match what was shown on the https://gamepad-tester.com/ website _(noted in comment in notebook code)_

Also, I'm a little confused because you mention 2 different browsers... could you elaborate in more detail the sequence of events you went through? Otherwise, why would you use chromium, when the game pad works on firefox?

thesaltymaker, 2021-11-24:
Thanks,

The Index of the controller is 0 on gampad-tester, and set to 0 in the jupyter notebook.  The error is "This browser does not support gamepads."

[image]

I followed the instructions in the teleoperation lab. 

I tried a different browser out of desperation to get the lab working after firefox showed the gampad was working on gamepad-tester, but not supported in the notebook.

santaimpersonator, 2021-11-29:
I have an Ubuntu (Linux) box at the office, which I can try the example with tomorrow.

Maybe try checking that the buttons also work on the website. If I remember correctly, moving the joysticks caused the `AXIS #` numbers to change and pressing the buttons cause the `B#` values to read `1`. If the values don't change, you might just need to try a different controller.

santaimpersonator, 2021-12-01:
Well, the example worked perfectly fine for me with Firefox on an Ubuntu PC. I tried with 2 different 8bitdo controllers with a Bluetooth USB dongle:

* https://www.sparkfun.com/products/17266
* https://www.sparkfun.com/products/17264
* https://www.sparkfun.com/products/17598

Unfortunately, I left my [Logitech USB controller](https://www.logitechg.com/en-us/products/gamepads/f310-gamepad.940-000110.html) at home... but my guess is that would've worked as well. I've never had any issue with it before on any other system (Windows, Mac, Pi, microcontroller, or gaming console).

*Oops... I forgot to check the widget version and I already left the office. I probably should have written down the Ubuntu and Firefox versions as well.*

I did notice that when I executed the code cell for the controller, the circle in the upper corner, next to the `Python 3`, was flashing. Once it connected, it would flash when any of the buttons were pressed. It probably just indicates when iPython (I am pretty sure that is what is used) is running in the background, but it might be another thing to check.

* Actually, that might be another way to also debug the issue. Use the terminal in the jupyter notebook (I don't think SSH-ing into the jetson will work since the code relies on the web browser _plug-in(?)_) and execute the code line by line with iPython3. There might be an error message that the Jupyter notebook isn't displaying (I've had that problem before... super annoying).

Otherwise, maybe try a different controller???

closed 2022-07-18 by jaybdub

ingowald, 2023-07-24:
Just in case anybody else runs into this: I had the same problem - gamepad-tester recognized the (android-)gamepad just fine, but jetbot's jupyternotebook errored out with "this browser does not support gamepads". Tried all of the above, didn't help; but switching to chrome instead of firefox made it work.