NVIDIA-AI-IOT/jetbot #498: Can we run examples without user login

mkhansenbot, 2021-11-18:
I'm trying to use the road follower example for a demo, but need to run it as a script without user input. I tried exporting the script, installing the dependencies (torch, etc), but when I run it, it segfaults at the torch to TRT conversion step.

Can you provide a non-jupyter python3 command line script?

jaybdub, 2021-11-18:
Hi @mkhansenbot ,

Thanks for reaching out!

Do you mind sharing the versions of software you're using (notably, JetPack, TensorRT, PyTorch), as well as the steps you took to convert the model to TensorRT.  

Any code snippets you have may be helpful :)

Best,
John

mkhansenbot, 2021-11-18:
Here's the failing code: I added the print statements
```
#!/usr/bin/env python3
# coding: utf-8

# # Road Following - Live demo (TensorRT)

# In this notebook, we will use model we trained to move JetBot smoothly on track.

import torch
device = torch.device('cuda')

# Load the TRT optimized model by executing the cell below

import torch
from torch2trt import TRTModule

model_trt = TRTModule()
print('Loading Tensor RT model')
model_trt.load_state_dict(torch.load('/home/jetbot/demos_ws/best_steering_model_xy_trt.pth'))
print('Model loaded')
```
I see this segfault:
```
(base) jetbot@jetbot:~/demo_ws$ python3.6 demo.py
Loading Tensor RT model
Segmentation fault (core dumped)
(base) jetbot@jetbot:~/demo_ws$
```
I'm running python3.6, torch 1.10. 

I don't know the versions of the others you mentioned. I had a hard time installing everything, if you have a container or instructions for installing the dependencies outside of Jupyter, that would help.

jaybdub, 2021-11-18:
Thanks for providing this information.

Regarding container, you could try the L4T PyTorch containers if it helps, they include most dependencies you'll need.  

https://catalog.ngc.nvidia.com/orgs/nvidia/containers/l4t-pytorch

After that, you just need to install torch2trt inside the container 

```bash
git clone https://github.com/NVIDIA-AI-IOT/torch2trt
cd torch2trt
python3 setup.py install
```

As for the issue you've encountered, I see that you're loading a model that's already been optimized.

```python
model_trt.load_state_dict(torch.load('/home/jetbot/demos_ws/best_steering_model_xy_trt.pth'))
```

Did you generate this TensorRT model in the same environment, or was it perhaps done on a different platform or with different version of software?  I believe if there is a platform or version mismatch, it can lead to segmentation fault.

Best,
John

mkhansenbot, 2021-11-18:
I generated the model in the Jupyter notebook. I'm just trying to run it on the same system outside of Jupyter. How would you recommend I do that?

mkhansenbot, 2022-01-11:
I never succeeded in getting this to work, is anyone able to provide a working python script?