NVIDIA-AI-IOT/jetbot #500: Incompatible keys issue in Road Following example

himadrir, 2021-11-25:
I've followed the steps to do the road following example, however I get this error

IncompatibleKeys(missing_keys=[], unexpected_keys=[])

when i run this line:
model.load_state_dict(torch.load('best_steering_model_xy.pth')) in live_demo file

can someone guide me what's happening?

jaybdub, 2021-12-14:
Hi @himadrir ,

Thanks for reaching out!

Usually this error indicates that the PyTorch code for the model weights that you're attempting to load don't match.

This could potentially happen if during training you call this line, but when loading the weights you don't

```python
model.fc = torch.nn.Linear(512, 2)
```

Does this error include any additional information?  

Best,
John

himadrir, 2021-12-30:
I am using the Jupyter example given with the jetbot, but AFAIK I see the line in both training and while loading as well. The problem still persists.