NVIDIA-AI-IOT/jetbot #507: Two libgomp in jetbot, but ONE is preloaded using LD_PRELOAD

jiapei100, 2021-12-23:
Hi, all:

I got the following error messages, one at a time... 

- `OSError: /home/jetbot/.local/lib/python3.6/site-packages/torch/lib/libgomp-d22c30c5.so.1: cannot allocate memory in static TLS block`
- `OSError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block`

There are 2 libgomp in my jetbot... It seems I shouldn't do `pip install torch` ? Then, should I build **torch** from source ?

Looking forward to your help...
Cheers

Tiryoh, 2022-01-20:
Hi @jiapei100,

It is addressed in https://github.com/rt-net/jnm_jupyternotebook/issues/14#issuecomment-968562735.

I guess modifying `/usr/local/share/jupyter/kernels/python3/kernel.json` with this patch will solve the problem.
https://github.com/rt-net/jnmouse_utils/blob/master/setup-scripts/src/jupyter.patch

jiapei100, 2022-01-20:
@Tiryoh 

Okay, I found it...
Happen to realize for **jetbot**, under folder `/usr/lib/aarch64-linux-gnu`,  **libgomp.so** is linked to **libomp.so.5**, as:

```console
➜  aarch64-linux-gnu pwd
/usr/lib/aarch64-linux-gnu
➜  aarch64-linux-gnu ll libgomp*
lrwxrwxrwx 1 root root   11 Jan  7  2018 libgomp.so -> libomp.so.5
lrwxrwxrwx 1 root root   16 Mar 10  2020 libgomp.so.1 -> libgomp.so.1.0.0
-rw-r--r-- 1 root root 180K Mar 10  2020 libgomp.so.1.0.0
```

What we need to do might be re-symbolic-link **libgomp.so** as:
➜  aarch64-linux-gnu ll libgomp*
lrwxrwxrwx 1 root root   12 Jan 20 12:03 libgomp.so -> libgomp.so.1
lrwxrwxrwx 1 root root   16 Mar 10  2020 libgomp.so.1 -> libgomp.so.1.0.0
-rw-r--r-- 1 root root 180K Mar 10  2020 libgomp.so.1.0.0
```

Testing now...

Tiryoh, 2022-01-21:
It is similar, but in my environment (JetPack 4.5.1), there was no `libgomp.so -> libomp.so.5`.

```
$ ls -l --time-style=iso /usr/lib/aarch64-linux-gnu/libgomp*
lrwxrwxrwx 1 root root     16 2020-03-10  /usr/lib/aarch64-linux-gnu/libgomp.so.1 -> libgomp.so.1.0.0
-rw-r--r-- 1 root root 183384 2020-03-10  /usr/lib/aarch64-linux-gnu/libgomp.so.1.0.0

$ dpkg -l | grep libgomp
ii  libgomp1:arm64                                8.4.0-1ubuntu1~18.04                             arm64        GCC OpenMP (GOMP) support library
```