NVIDIA-AI-IOT/jetbot #509: The kernel appears to have died. It will restart automatically.

jiapei100, 2021-12-26:
I'm running JetBot on a Jetson Nano 2G. For Example 3 - Collision avoidance (Please refer to [JetBot Wiki Examples](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/examples) ), after I successfully running the first 4 cells, when I tried to run cell 5:

```console
import traitlets
from IPython.display import display
import ipywidgets.widgets as widgets
from jetbot import Camera, bgr8_to_jpeg

camera = Camera.instance(width=224, height=224)
image = widgets.Image(format='jpeg', width=224, height=224)
blocked_slider = widgets.FloatSlider(description='blocked', min=0.0, max=1.0, orientation='vertical')
speed_slider = widgets.FloatSlider(description='speed', min=0.0, max=0.5, value=0.0, step=0.01, orientation='horizontal')

camera_link = traitlets.dlink((camera, 'value'), (image, 'value'), transform=bgr8_to_jpeg)

display(widgets.VBox([widgets.HBox([image, blocked_slider]), speed_slider]))
```

A dialog jumped out saying **The kernel appears to have died. It will restart automatically.**. 

How much memory does it require to run this **collision avoidance example**?

Cheers

santaimpersonator, 2022-01-05:
The example needs almost all of the available memory, which is why it crashes. I don't remember all the steps that I did, but you need to at least disable the desktop GUI.

jiapei100, 2022-01-05:
@santaimpersonator Thank you... 
Do you mean the GUI on jetbot? or the GUI on my remote desktop?
I‘er already disabled the GUI on jetbot.

santaimpersonator, 2022-01-05:
The GUI on the Jetbot needs to be disabled. I have some notes here: https://github.com/NVIDIA-AI-IOT/jetbot/issues/428#issuecomment-931599482

I remember it took several hours for the example to start running.

closed 2022-07-15 by jaybdub