NVIDIA-AI-IOT/jetbot #510: Waveshare Jetbot with Jetson Nano 4GB set-up errors

CyberPhysicalSystemsRobotics, 2022-01-04:
Hello, I have a new waveshare Jetbot with Jetson Nano 4GB but having issues on set-up with a number of errors on boot up and the OLED does not display probably due to the Wi-Fi errors on boot up see attached picture.

I have used the Waveshare image link on their wiki site but aware on the other Jetbot site there is separate Nano 2GB and 4GB image versions..

Has anyone seen this errors messages in the attached picture or can offer any advice?

Many Thanks [image]

santaimpersonator, 2022-01-05:
_*I'm not familiar with the Waveshare Jetbot images_

The second line, just means it can't detect the camera. At the 6.4 mark, it looks like you might be running in low power mode... If I remember correctly, it takes a while for the Jetson to boot in that configuration.

Otherwise, if you have updated the firmware (with the Jetpack 4.5 image), I believe certain, older Jetpack based images become incompatible.

CyberPhysicalSystemsRobotics, 2022-01-05:
Quick update Jan 5th 

To try rule out issues I have tried the Jetson original SD image OShttps://developer.nvidia.com/jetson-nano-sd-card-image I have unconnected the Jetbot and the Jetson so the Jetson is stand alone[image].  I have also flashed the image to the SD with the etcher application and also the Win32DiskImager.exe application with the following errors at boot up some the same as with the Jetbot Image.
Image.jpegImage.jpeg
There looks like 6 errors common with the Jetbot Image which will not be helping, have you seen these before and could recommend what else I could test/ rectify?

santaimpersonator, 2022-01-05:
_*It looks like you are missing a few images from your last comment._

I would recommend using the [Nvidia forum](https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-nano/76) to troubleshoot your issue with the Jetson board. This repo has only only two primary maintainers from Nvidia. You'll probably have more available resources, by posting on the Nvidia developer forum.

That said, I would reiterate that it takes a while for the board to boot in the low power mode, since it uses less CPU cores and at ~1/2 the speed. Otherwise, you are using an image that is incompatible with the firmware on the board. If those aren't your issues, you will probably want to go to the forum for help or read the developer documentation for re-configuring the Jetson module with the SDK manager _(Linux computer required)_.

closed 2022-01-15 by CyberPhysicalSystemsRobotics