NVIDIA-AI-IOT/jetbot #513: Wave share Jetbot Other Code functions..

CyberPhysicalSystemsRobotics, 2022-01-15:
Hello, 

I have Waveshare Jetbot but looking for more info beyond info online...has anyone any info on the following:

1. The example Jupiter notebooks only have few command, where can I find a list off all Jetbot python commands? 
2. Is the a Jetbot software Architecture diagram?
Does the Jetbot use Ros 1 or Ros 2?
3. Where can I find a list of all ROS topics?

Thanks Steve

tokk-nv, 2022-02-01:
Hi CyberPhysicalSystemsRobotics,

1. `basic_motion.ipynb` demonstrates all the methods defined under `Robot` class.
2. There is no architecture diagram
3. JetBot is not based on ROS or ROS 2.
4. n/a

brianlmerritt, 2022-04-04:
If you want ROS or ROS2, look at 

https://developer.nvidia.com/blog/implementing-robotics-applications-with-ros-2-and-ai-on-jetson-platform-2/

Note a 2gb nano probably won't have enough memory to run desktop version, rviz, gazebo etc

I haven't tried the above, just on my todo list.

closed 2022-07-18 by jaybdub