NVIDIA-AI-IOT/jetbot #525: Cannot access JetBot via web

Anonymou567, 2022-03-14:
Good Day,

After connecting to Wi-Fi and starting up the JetBot. My JetBot OLED is not showing any IP and i am unable to connect to my jetbot via web browser as it just shows "This site can't be reached". I am currently using the Jetson Nano 2GB.

[image]

chitoku, 2022-03-16:
Hi,

You may want to connect a USB cable from your PC to the micro-B port of Jetson Nano, and check if you can login to the terminal.
Something like this.
https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/software_setup.md#step-2---power-on-and-connect-over-usb

Anonymou567, 2022-03-17:
Hi,

Thanks for the reply. I have solved the issue by flashing an older image. v0.4.3 didn't work but v0.4.2 worked.

closed 2022-03-17 by Anonymou567