NVIDIA-AI-IOT/jetbot #532: network-manager package not installed on jupyterlab server

lucabrack, 2022-04-04:
In order to set up the Jetbot I have to connect it to a monitor and a keyboard (Following this [instruction](https://jetbot.org/master/software_setup/sd_card.html)).

But without any preparation I can connect the Jetbot to my computer via USB-Cable and access the jupyterlab server with the IP-Adress `http://192.168.55.1:8888/`. I tried to connect to my WiFi over this connection, but the network-manager package used in the above mentioned instruction is not installed. Installing it is not possible, because because of the missing internet connection.

If I could connect to WiFi over the USB-Cable I wouldn't need an extra monitor.

Is there another way to connect to the wifi without `nmcli` or is it possible to provide an image who has the network-manager package installed?

naisy, 2022-04-05:
The JupyterLab of JetBot JetPack 4.5 image is running on a docker container.
The docker container usually does not have permission to access reboot, shutdown, systemctrl, service and nmcli.

## Way 1
Probably the easiest way is to install JupyterLab on localhost and run it with a different port number such as 8889.
But, it doesn't make sense to have different port numbers.

## Way 2
You can install jetbot directly on localhost. But it is high cost for maintainers. If the maintainer finds a problem with the package during installation, he will have to redo the flash of the jetpack image.

## Way 3
So, using docker containers, I tried to find out if these are technically feasible.
The result is that these are accessible from the docker container.
No network-manager or other software is installed.
They must be mounted and used from localhost.

But, still problem exists.
It takes about 30 seconds for a normal ssh login.
If the /sbin/init container is stopped, ssh login will return to normal speed.
However, in order to run the service from the JupyterLab container, it must be started once per reboot.
Hopefully a solution to this problem will fall from the sky.

### Way 3 mytest
Jetson Nano 2GB/4GB / JetPack 4.6.1 / 64GB sdcard

`vi run-jetbot.sh`
```
#!/bin/bash
CSI_CAMERA=/tmp/argus_socket

########################################
# docker image
########################################
IMG=faborobot/jetbot-jp461

########################################
# JetBot settings
########################################
# OLED DISLPAY
docker run \
    --restart always \
    -it -d \
    --name=jetbot_ipaddress \
    --privileged \
    --network=host \
$IMG \
python3 ~/stats.py

# INIT
docker run \
    --runtime nvidia \
    --restart always \
    --memory-swap=-1 \
    -it -d \
    -e SHELL=/bin/bash \
    -e OPENBLAS_CORETYPE=ARMV8 \
    -v /var/run/dbus/system_bus_socket:/var/run/dbus/system_bus_socket:ro \
    -v /etc/localtime:/etc/localtime:ro \
    -v $CSI_CAMERA:$CSI_CAMERA \
    -v /dev/:/dev/ \
    -v /sbin/shutdown:/sbin/shutdown:ro \
    -v /sbin/reboot:/sbin/reboot:ro \
    -v /bin/systemctl:/bin/systemctl:ro \
    -v /lib/systemd/libsystemd-shared-237.so:/lib/systemd/libsystemd-shared-237.so:ro \
    -v /lib/aarch64-linux-gnu/libcryptsetup.so.12:/lib/aarch64-linux-gnu/libcryptsetup.so.12:ro \
    -v /lib/aarch64-linux-gnu/libidn.so.11:/lib/aarch64-linux-gnu/libidn.so.11:ro \
    -v /lib/aarch64-linux-gnu/libjson-c.so.3:/lib/aarch64-linux-gnu/libjson-c.so.3:ro \
    -v /usr/lib/aarch64-linux-gnu/libip4tc.so.0:/usr/lib/aarch64-linux-gnu/libip4tc.so.0:ro \
    -v /usr/lib/aarch64-linux-gnu/libargon2.so.0:/usr/lib/aarch64-linux-gnu/libargon2.so.0:ro \
    -v /usr/lib/aarch64-linux-gnu/libnm.so.0:/usr/lib/aarch64-linux-gnu/libnm.so.0:ro \
    -v /usr/lib/aarch64-linux-gnu/libpolkit-agent-1.so.0:/usr/lib/aarch64-linux-gnu/libpolkit-agent-1.so.0:ro \
    -v /usr/lib/aarch64-linux-gnu/libpolkit-gobject-1.so.0:/usr/lib/aarch64-linux-gnu/libpolkit-gobject-1.so.0:ro \
    -v /sbin/:/sbin/:ro \
    -v /usr/sbin/:/usr/sbin/:ro \
    -v /sbin/init:/sbin/init:ro \
    -v /usr/lib/NetworkManage/:/usr/lib/NetworkManage/:ro \
    -v /run/:/run/ \
    -v /usr/bin/nmcli:/usr/bin/nmcli:ro \
    --privileged \
    --network=host \
$IMG \
/sbin/init

# JupyterLab
docker run \
    --runtime nvidia \
    --restart always \
    --memory-swap=-1 \
    -it -d \
    --name=jetbot_jupyter \
    -e SHELL=/bin/bash \
    -e OPENBLAS_CORETYPE=ARMV8 \
    -v /var/run/dbus/system_bus_socket:/var/run/dbus/system_bus_socket:ro \
    -v /etc/localtime:/etc/localtime:ro \
    -v $CSI_CAMERA:$CSI_CAMERA \
    -v /dev/:/dev/ \
    -v /sbin/shutdown:/sbin/shutdown:ro \
    -v /sbin/reboot:/sbin/reboot:ro \
    -v /bin/systemctl:/bin/systemctl:ro \
    -v /lib/systemd/libsystemd-shared-237.so:/lib/systemd/libsystemd-shared-237.so:ro \
    -v /lib/aarch64-linux-gnu/libcryptsetup.so.12:/lib/aarch64-linux-gnu/libcryptsetup.so.12:ro \
    -v /lib/aarch64-linux-gnu/libidn.so.11:/lib/aarch64-linux-gnu/libidn.so.11:ro \
    -v /lib/aarch64-linux-gnu/libjson-c.so.3:/lib/aarch64-linux-gnu/libjson-c.so.3:ro \
    -v /usr/lib/aarch64-linux-gnu/libip4tc.so.0:/usr/lib/aarch64-linux-gnu/libip4tc.so.0:ro \
    -v /usr/lib/aarch64-linux-gnu/libargon2.so.0:/usr/lib/aarch64-linux-gnu/libargon2.so.0:ro \
    -v /usr/lib/aarch64-linux-gnu/libnm.so.0:/usr/lib/aarch64-linux-gnu/libnm.so.0:ro \
    -v /usr/lib/aarch64-linux-gnu/libpolkit-agent-1.so.0:/usr/lib/aarch64-linux-gnu/libpolkit-agent-1.so.0:ro \
    -v /usr/lib/aarch64-linux-gnu/libpolkit-gobject-1.so.0:/usr/lib/aarch64-linux-gnu/libpolkit-gobject-1.so.0:ro \
    -v /sbin/:/sbin/:ro \
    -v /usr/sbin/:/usr/sbin/:ro \
    -v /sbin/init:/sbin/init:ro \
    -v /usr/lib/NetworkManage/:/usr/lib/NetworkManage/:ro \
    -v /run/:/run/ \
    -v /usr/bin/nmcli:/usr/bin/nmcli:ro \
    --privileged \
    --network=host \
$IMG \
jupyter lab --ip=0.0.0.0 --no-browser --notebook-dir=/root/Notebooks --allow-root
```
```
chmod 755 run-jetbot.sh
sudo su
./run-jetbot.sh
```
Open JupyterLab Terminal
```
systemctl status NetworkManager.service
nmcli connection
nmcli device wifi connect 'MYSSID' password 'MyPASSWORD' ifname wlan0
nmcli connection
# nmcli connection delete UUID # if you want to delete wifi connection
```

naisy, 2022-05-13:
@lucabrack 
I noticed that there are Docker Tips in the documentation.
You may want to use this.
1. ssh login to jetson localhost from jupyterlab terminal.
2. Run the wifi configuration command.

https://jetbot.org/master/reference/docker_tips.html

1. ssh login
```
ssh jetbot@0.0.0.0
```
2. wifi configuration
```
sudo nmcli device wifi connect 'MySSID' password 'MyPassword' ifname wlan0
```

closed 2022-07-18 by jaybdub