NVIDIA-AI-IOT/jetbot #533: basic motion do not work correctly

ahforoughi, 2022-04-09:
Hi,
I have done everything according to the hardware setup section but I have used different motor drivers. (this model: TB6612 CJMCU)
Everything works well but when I use the basic motion notebook to move jetbot, the jetbot moves in the wrong direction. For example when I run ``` robot.left(speed=0.5)``` it goes forward. NO directions are working properly. 
I check the solders and circuits once again and it seems are all ok.
(I am using jetson nano 2GB)
Does anyone have any idea how can I fix this?

naisy, 2022-04-12:
@ahforoughi 
Let's see the robot.py code.
https://github.com/NVIDIA-AI-IOT/jetbot/blob/636add1/jetbot/robot.py#L78
This sets the left motor to negative speed and the right motor to positive speed.
```
def left(self, speed=1.0):
    self.left_motor.value = -speed
    self.right_motor.value = speed
```

In your case:
```
robot.left(speed=0.5)
```
Moving forward means that the +- of the left motor is reversed, or the left motor is installed in the opposite direction.

Swap the +- wiring of the left motor or swap the direction of the left motor.

ahforoughi, 2022-04-13:
> @ahforoughi Let's see the robot.py code. https://github.com/NVIDIA-AI-IOT/jetbot/blob/636add1/jetbot/robot.py#L78 This sets the left motor to negative speed and the right motor to positive speed. [...]
@naisy 
Thanks, but I tried this, which did not affect the motor's rotation. 
I have changed the code in jetbot dir, and restarted the images during the software setup once more. 
Do we have to build the images from scratch?

ahforoughi, 2022-04-16:
>

ahforoughi, 2022-04-16:
> @ahforoughi
> 
> > Do we have to build the images from scratch?
> 
> I think, no need.
> 
> > Thanks, but I tried this, which did not affect the motor's rotation.
> 
> It's strange. I want to understand this correctly. If you swap the plus and minus, the motor will rotate in the opposite direction. Are you using special motors?
> 
> In basic motion, you can also operate the right, forward, and backward buttons. Do these all move forward?

No, it's not special motors. it is the same as the instruction.
and when the left command executes, it goes forward. when the forward command executes it goes to the right. it is like all of the directions are not working correctly.

closed 2022-04-16 by ahforoughi

naisy, 2022-04-17:
@ahforoughi 

Change the wiring.
before:
[image]
after:
[image]

> when the left command executes, it goes forward. when the forward command executes it goes to the right.

In the figure, you can see that the left and right motor connections are reversed.
After that, reverse the plus and minus of the right motor to make the correct connection.
[image]