NVIDIA-AI-IOT/jetbot #536: JetBot Camera keeps rebooting Jetson Nano

Kivin1, 2022-05-18:
Hi

I am facing one issue. i am executing data collection script in collision avoidance,. Camera.instance() give me no response and after a while, the jetson nano will reboot. After rebooting, the issue still remain the same

The camera is working well when running nvgstcapture-1.0

Could you please help to check on this issue.

Matthew-Hu-cmd, 2022-05-29:
I'm facing the same issue here, while running the veido proccing node along with other ROS nodes, it will reboot after a while...
but it worked like a charm if I only turn the navigation nodes up...

Juff-Ma, 2022-06-17:
me to, please help. GSTream prints producer connected and the jetbot reboots

Kivin1, 2022-06-20:
I am using jetcam as a workaround.. it works to me

Juff-Ma, 2022-06-20:
Could you please give me a tutorial for this or example?

jaybdub, 2022-07-15:
You can find the JetCam package and instructions [here](https://github.com/NVIDIA-AI-IOT/jetcam) 

Hope this helps.  Going to close for now.  Feel free to re-open if you run into any issues or have questions.

Best,
John

closed 2022-07-15 by jaybdub