NVIDIA-AI-IOT/jetbot #537: It is not booting. No GUI Display

anandbharadwaj1, 2022-06-01:
[image]

ThaiVuz, 2022-06-03:
i also have a problem like you ? If you know how to fix it please help me !!!!

jaybdub, 2022-07-15:
Hi All,

Thansk for reaching out.

To conserve memory consumption, we disable the GUI on JetBot by default (since it's programmed over a Jupyter Web server).

If you want to re-enable the GUI, you can execute the following command.

```bash
sudo systemctl set-default graphical.target
```

You can find more details on [this page](http://jetbot.org/master/software_setup/docker.html).

Hope this helps, please let me know if you have any questions or run into any issues.

Best,
John

closed 2022-07-15 by jaybdub