NVIDIA-AI-IOT/jetbot #542: Jetbot motor test

anandbharadwaj1, 2022-06-13:
We were able to run the motor.ipynb script from Jupiter notebooks. We would like to create a python file and run the motors and link with the other part of project which involves subscribing to data. Could you tell me how to create a python script for running the motors and run it using the command python motor.py.

naisy, 2022-06-14:
Please refer to basic_motion.

a.py
```python
from jetbot import Robot
import time

robot = Robot()
robot.left(speed=0.3)
time.sleep(0.5)
robot.stop()
```
run
```
python3 a.py
```

anandbharadwaj1, 2022-06-17:
Tried this but there is some issue with the i2c bus module

naisy, 2022-06-17:
Is jetbot installed correctly?
Do you have error log?

anandbharadwaj1, 2022-06-20:
Yes the installation of jetbot is done properly. We used a prebuilt jetbot image while flashing the SD card

naisy, 2022-06-20:
Can you past these two logs?

1. check jetbot installed or not
```
pip3 list installed | grep jetbot
```

2. python error log
```
python3 -c 'from jetbot import Robot'
```

closed 2022-07-18 by jaybdub