NVIDIA-AI-IOT/jetbot #546: How to access my jetbot’s csi camera(IMX 219-160 camera) using opencv-python

mmzz16, 2022-07-14:
hello, so i wanted to access my CSI camera using OpenCV because I am working on a line following robot project. can anyone help?

jaybdub, 2022-07-15:
Hi mmzz16,

Thanks for reaching out!

I'll assume you're using Raspberry Pi Camera V2 (or similar IMX219 camera).  In that case,

```python
import cv2

camera_pipeline = "nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=640, height=480, format=(string)NV12, framerate=(fraction)30/1 ! nvvidconv ! video/x-raw, width=(int)640, height=(int)480, format=(string)BGRx ! videoconvert ! appsink"

camera = cv2.VideoCapture(camera_pipeline, cv2.CAP_GSTREAMER)

re, image = camera.read()

```

This is what's done inside of [JetCam](https://github.com/NVIDIA-AI-IOT/jetcam).

Or if you find the need to optimize your application further, @dusty-nv has created an efficient version that doesn't depend on OpenCV in [jetson-utils](https://github.com/dusty-nv/jetson-utils) and allows you to quickly acquire images to a GPU buffer for processing.

Hope this helps, let me know if you run into any issues or have any questions.

Best,
John

closed 2022-07-15 by jaybdub

mmzz16, 2022-07-19:
Dear John,

when i try to run the code you wrote through ssh it returns the
following error can you help please?
[image: image.png]