NVIDIA-AI-IOT/jetbot #547: Python Environment

NikolaAndro, 2022-07-18:
Can anyone explain why the python environment does not work as usual on the jetbot please?

I am trying to install  some packages such as numpy, but getting the following error:

```The directory '/home/jetbot/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/jetbot/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting numpy
  Downloading https://files.pythonhosted.org/packages/51/60/3f0fe5b7675a461d96b9d6729beecd3532565743278a9c3fe6dd09697fa7/numpy-1.19.5.zip (7.3MB)
    100% |████████████████████████████████| 7.3MB 45kB/s 
Installing collected packages: numpy
  Running setup.py install for numpy ... error
    Complete output from command /home/jetbot/lane_following/venv_lane/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-hmi6qgd1/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-t20lwecp-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/jetbot/lane_following/venv_lane/include/site/python3.6/numpy:
    Running from numpy source directory.
    
    Note: if you need reliable uninstall behavior, then install
    with pip instead of using `setup.py install`:
    
      - `pip install .`       (from a git repo or downloaded source
                               release)
      - `pip install numpy`   (last NumPy release on PyPi)
    
    
    Cythonizing sources
    Processing numpy/random/_bounded_integers.pxd.in
    Processing numpy/random/bit_generator.pyx
    Traceback (most recent call last):
      File "/tmp/pip-build-hmi6qgd1/numpy/tools/cythonize.py", line 59, in process_pyx
        from Cython.Compiler.Version import version as cython_version
    ModuleNotFoundError: No module named 'Cython'
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/tmp/pip-build-hmi6qgd1/numpy/tools/cythonize.py", line 235, in <module>
        main()
      File "/tmp/pip-build-hmi6qgd1/numpy/tools/cythonize.py", line 231, in main
        find_process_files(root_dir)
      File "/tmp/pip-build-hmi6qgd1/numpy/tools/cythonize.py", line 222, in find_process_files
        process(root_dir, fromfile, tofile, function, hash_db)
      File "/tmp/pip-build-hmi6qgd1/numpy/tools/cythonize.py", line 188, in process
        processor_function(fromfile, tofile)
      File "/tmp/pip-build-hmi6qgd1/numpy/tools/cythonize.py", line 64, in process_pyx
        raise OSError('Cython needs to be installed in Python as a module')
    OSError: Cython needs to be installed in Python as a module
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-hmi6qgd1/numpy/setup.py", line 508, in <module>
        setup_package()
      File "/tmp/pip-build-hmi6qgd1/numpy/setup.py", line 488, in setup_package
        generate_cython()
      File "/tmp/pip-build-hmi6qgd1/numpy/setup.py", line 285, in generate_cython
        raise RuntimeError("Running cythonize failed!")
    RuntimeError: Running cythonize failed!
    
    ----------------------------------------
Command "/home/jetbot/lane_following/venv_lane/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-hmi6qgd1/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-t20lwecp-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/jetbot/lane_following/venv_lane/include/site/python3.6/numpy" failed with error code 1 in /tmp/pip-build-hmi6qgd1/numpy/ ```

I am also having issues with installing scipy.

Thank you!

jaybdub, 2022-07-19:
Hi @NikolaAndro ,

Thanks for reaching out!

According to the error log, it seems like Cython isn't installed on your system.

Could you try the following and see if it helps?

```bash
sudo python3 -m pip install Cython
```

Let me know if this helps or you have any questions.

Best,
John

NikolaAndro, 2022-07-19:
Hello @jaybdub ,

Thanks for the quick response! 

I installed it! Now, I am probably not able to use it in a virtual environment since it is installed on the system itself? 

I found [this tutorial](https://pyimagesearch.com/2020/03/25/how-to-configure-your-nvidia-jetson-nano-for-computer-vision-and-deep-learning/) for virtual environment on jetbot. Is this the way to go about it?

Thanks,
Nikola

jaybdub, 2022-07-19:
I'm not too familiar with virtual environments in Python, but I would imagine it would just require installing Cython inside the virtual environment.

Perhaps using something like [this](https://packaging.python.org/en/latest/guides/installing-using-pip-and-virtual-environments/).

Activate the virtual environment, install requirements.

Let me know if this helps or you have any questions.

Best,
John

NikolaAndro, 2022-07-19:
I will keep this issue opened for a bit and post a solution once I find it.

NikolaAndro, 2022-08-23:
I can confirm that the tutorial I mentioned above is working. There were some errors, but manageable.

closed 2022-08-23 by NikolaAndro

sriharish-324, 2023-11-28:
Hello, I need to install Mediapipe for my jetbot Which is using **Python 3.6.9** which does not supports mediapipe.