NVIDIA-AI-IOT/jetbot #549: Wiki access request for dead links and 3D printing section update.

danielwilczak101, 2022-07-30:
Hello NVIDIA jet bot community,

I was wondering if I could get access to update the wiki pages to fix all the dead links and just give it a general update with the 3D printing section.

I've built build previous wiki's for my own python packages. As you can see in the link below.
https://github.com/danielwilczak101/EasyGA/wiki

Thank you,
Daniel W

jaybdub, 2022-08-01:
Hi @danielwilczak101 ,

Thanks for reaching out!

We've actually migrated away from the Wiki in favor of Mkdocs for generating documentation.

For reference, the 3D printing page source is now located here

https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/docs/3d_printing.md

Some instructions for working with the new documentation are located here

https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/CONTRIBUTING.md#deploying

These documentation pages are automatically built and hosted on our GitHub pages documentation

http://jetbot.org/master/

Do you still see the dead links in the newer documentation?

Best,
John

danielwilczak101, 2022-08-02:
Hey @jaybdub 

This is awesome. I love that the docs have been moved into the repository. Way better then using the Wiki.

The dead links are still there but I'll make a pull request with updated links and a prototype 3d printing page that hopefully helps new users more then the current.

Thank you,
Daniel

closed 2022-08-02 by danielwilczak101