NVIDIA-AI-IOT/jetbot #550: Use H bridge on jetbot

irvinmarceloc, 2022-08-04:
Hi guys. I am trying to use an H-bridge on the jetbot however when running the basic moves notebook I get the following error.
What modifications do I have to make to be able to use an H-bridge?

OSErrorTraceback (most recent call last)
<ipython-input-32-b418ad9f6ab3> in <module>
----> 1 robot = Robot()

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.3-py3.6.egg/jetbot/robot.py in __init__(self, *args, **kwargs)
     20     def __init__(self, *args, **kwargs):
     21         super(Robot, self).__init__(*args, **kwargs)
---> 22         self.motor_driver = Adafruit_MotorHAT(i2c_bus=self.i2c_bus)
     23         self.left_motor = Motor(self.motor_driver, channel=self.left_motor_channel, alpha=self.left_motor_alpha)
     24         self.right_motor = Motor(self.motor_driver, channel=self.right_motor_channel, alpha=self.right_motor_alpha)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py in __init__(self, addr, freq, i2c, i2c_bus)
    229         self.motors = [ Adafruit_DCMotor(self, m) for m in range(4) ]
    230         self.steppers = [ Adafruit_StepperMotor(self, 1), Adafruit_StepperMotor(self, 2) ]
--> 231         self._pwm = PWM(addr, debug=False, i2c=i2c, i2c_bus=i2c_bus)
    232         self._pwm.setPWMFreq(self._frequency)
    233 

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in __init__(self, address, debug, i2c, i2c_bus)
     57         self.i2c = get_i2c_device(address, i2c, i2c_bus)
     58         logger.debug("Reseting PCA9685 MODE1 (without SLEEP) and MODE2")
---> 59         self.setAllPWM(0, 0)
     60         self.i2c.write8(self.__MODE2, self.__OUTDRV)
     61         self.i2c.write8(self.__MODE1, self.__ALLCALL)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in setAllPWM(self, on, off)
     93     def setAllPWM(self, on, off):
     94         "Sets a all PWM channels"
---> 95         self.i2c.write8(self.__ALL_LED_ON_L, on & 0xFF)
     96         self.i2c.write8(self.__ALL_LED_ON_H, on >> 8)
     97         self.i2c.write8(self.__ALL_LED_OFF_L, off & 0xFF)

/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.4-py3.6.egg/Adafruit_GPIO/I2C.py in write8(self, register, value)
    114         """Write an 8-bit value to the specified register."""
    115         value = value & 0xFF
--> 116         self._bus.write_byte_data(self._address, register, value)
    117         self._logger.debug("Wrote 0x%02X to register 0x%02X",
    118                      value, register)

/usr/local/lib/python3.6/dist-packages/Adafruit_PureIO-1.1.8-py3.6.egg/Adafruit_PureIO/smbus.py in write_byte_data(self, addr, cmd, val)
    320         # Send the data to the device.
    321         self._select_device(addr)
--> 322         self._device.write(data)
    323 
    324     def write_word_data(self, addr, cmd, val):

OSError: [Errno 121] Remote I/O error

jaybdub, 2022-08-05:
Hi @irvinmarceloc ,

Thanks for reaching out!

Are you able to turn the motors without the JetBot software?

If so do you mind sharing the code to do so, and I could provide more detail guidance on how to implement your own Robot class.

Best,
John

irvinmarceloc, 2022-08-05:
> 

Hey Hello.
If I can spin the motors with an arduino.
The code that I am using is the one that comes in the image of the operating system for jetbot in its latest version.
It would be this basic move code.
https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/notebooks/basic_motion/basic_motion.ipynb
Which is in the repository.
From jupyter notebook I execute step by step and when I execute from jetbot import Robot that is where the problem is generated.
Greetings. I attach the code.

_BasicMotion
Welcome to JetBot's browser based programming interface! This document is called a Jupyter Notebook, which combines text, code, and graphic display all in one! Pretty neat, huh? If you're unfamiliar with Jupyter we suggest clicking the Help drop down menu in the top toolbar. This has useful references for programming with Jupyter.

In this notebook, we'll cover the basics of controlling JetBot.

Importing the Robot class
To get started programming JetBot, we'll need to import the Robot class. This class allows us to easily control the robot's motors! This is contained in the jetbot package.

If you're new to Python, a package is essentially a folder containing code files. These code files are called modules.

To import the Robot class, highlight the cell below and press ctrl + enter or the play icon above. This will execute the code contained in the cell

from jetbot import Robot
Now that we've imported the Robot class we can initialize the class instance as follows.

bot = bot()
commanding the robot
Now that we've created our Robot instance we named "robot", we can use this instance to control the robot. To make the robot spin counterclockwise at 30% of it's max speed we can call the following

WARNING: This next command will make the robot move! Please make sure the robot has clearance.

robot.left(speed=0.3)_

irvinmarceloc, 2022-08-05:
I simply installed the OS for jetbot in its latest version and made the following connections
[image]
[image]

danielwilczak101, 2022-08-06:
@irvinmarceloc 

Are you able to detect the address of the PCA9685 using i2c detect? I ask this because you're having a ``Remote I/O error``.

If you have never done this before. Just open terminal and install the i2c tool and use the second command to see if the address shows up. If it shows up then you know its not the connection. But something else.

```Bash
sudo apt-get install i2c-tools
i2cdetect -y 1
```

irvinmarceloc, 2022-08-07:
If it appears at address 27. Right now I don't have the jetson handy because it's in my workplace.
But the address that appears is 27.

danielwilczak101, 2022-08-08:
I would suggest trying to move the motor without using the jetbot code first. The jetbot library relies on the user using the adafruit motor controller because it was built using the adafruit's driver software. This is why a lot of the errors say ``Adafruit_PureIO-1.1.8-py3.6.egg/Adafruit_PureIO/...``.

(Other option) If you have it working with an Arduino, then you can always use i2c from the jetson to the Arduino to issue it commands to move.

https://dronebotworkshop.com/i2c-arduino-raspberry-pi/

This is a raspberry pi tutorial but i2c works the same for the Jetson.

irvinmarceloc, 2022-08-12:
Buenas tardes,
I actually gave up on that attempt. I think I'm going to buy the controller that it indicates in the list.
Thank you so much for everything

closed 2022-08-12 by irvinmarceloc