NVIDIA-AI-IOT/jetbot #552: Road following not yielding expected results

EnesiMe, 2022-08-11:
I'm not sure if it is my data collection method or the training but I keep getting the same erratic result with the road following live demo.  The robot steers off the lane and moves randomly. I followed the instructions in the data_collection.ipynb that utilizes the mouse click.  How do I go about solving this issue?

I also noticed there was a frame rate drop  after running the camera.observe(execute, names='value') cell, is this normal? 

Jetson Nano 4gb, 64gb card running jetbot primarily (not docker)

ken-9, 2022-09-08:
Hi, @EnesiMe.

It might be a good idea to collect data from various angles.
Also I had similar issue, I solved it by collecting data from various angles.

I recommend that you watch this video. 
This video explain method for robot`s good control. 

2:53~
[Jetson AI Fundamentals-S2E4-JetBot Road Following](https://youtu.be/VY7m4lPIOj0)

EnesiMe, 2022-09-19:
So I followed that from the start and i was able to confirm that the model was working, However, the motors do not respond accordingly. I have: 
- Adjusted the kd, steering gain, and motor speed
- Trained over  1000 images.
[Here](https://drive.google.com/file/d/1kxL1-ejDe5jtcRs0fVQgIeXFasODI21t/view?usp=sharing) is a link to a video demonstration of the model working after I disconnected the motors.
https://drive.google.com/file/d/1kxL1-ejDe5jtcRs0fVQgIeXFasODI21t/view?usp=sharing

Is there anyone else that had this issue? Is my data  at fault? @jaybdub

tevef, 2022-10-19:
Hello i started to use jetbot now but i dont find Road Following In the Notebook what have i to  do @ken-9 ken-9

ken-9, 2022-10-19:
@tevef 
Did you do step below?

> https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Software-Setup#step-5---install-latest-software-optional

David-VTUK, 2023-03-01:
@EnesiMe - Did you make any progress on this? I'm experiencing something similar.