NVIDIA-AI-IOT/jetbot #560: sudo python3 setup.py install  Error reported in the construction environment

Unlicensed-driver-ljx, 2022-09-15:
sudo python3 setup.py install
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ljx/jetbot
Scanning dependencies of target ssd_tensorrt
[ 50%] Building CXX object jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/FlattenConcat.cpp.o
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:113:9: error: ‘int FlattenConcat::enqueue(int, const void* const*, void**, void*, cudaStream_t)’ marked ‘override’, but does not override
     int enqueue (int batchSize, const  void* const* inputs, void** outputs, void*, cudaStream_t stream)  override
         ^~~~~~~
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:144:10: error: looser throw specifier for ‘virtual void FlattenConcat::serialize(void*) const’
     void serialize(void* buffer) const override
          ^~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from /home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:560:18: error:   overriding ‘virtual void nvinfer1::IPluginV2::serialize(void*) const noexcept’
     virtual void serialize(void* buffer) const noexcept = 0;
                  ^~~~~~~~~
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:168:10: error: looser throw specifier for ‘virtual void FlattenConcat::configureWithFormat(const Dims*, int, const Dims*, int, nvinfer1::DataType, nvinfer1::PluginFormat, int)’
     void configureWithFormat(const Dims* inputs, int nbInputs, const Dims* outputDims, int nbOutputs, nvinfer1::DataType type, nvinfer1::PluginFormat format, int maxBatchSize) override
          ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from /home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:503:18: error:   overriding ‘virtual void nvinfer1::IPluginV2::configureWithFormat(const Dims*, int32_t, const Dims*, int32_t, nvinfer1::DataType, nvinfer1::PluginFormat, int32_t) noexcept’
     virtual void configureWithFormat(Dims const* inputDims, int32_t nbInputs, Dims const* outputDims, int32_t nbOutputs,
                  ^~~~~~~~~~~~~~~~~~~
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:198:10: error: looser throw specifier for ‘virtual bool FlattenConcat::supportsFormat(nvinfer1::DataType, nvinfer1::PluginFormat) cons ’
     bool supportsFormat(DataType type, PluginFormat format) const override
          ^~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from /home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:478:18: error:   overriding ‘virtual bool nvinfer1::IPluginV2::supportsFormat(nvinfer1::DataType, nvinfer1::PluginFormat) const noexcep ’
     virtual bool supportsFormat(DataType type, PluginFormat format) const noexcept = 0;
                  ^~~~~~~~~~~~~~
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:203:17: error: looser throw specifier for ‘virtual const char* FlattenConcat::getPluginType() const’
     const char* getPluginType() const override { return FLATTENCONCAT_PLUGIN_NAME; }
                 ^~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from /home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:431:30: error:   overriding ‘virtual const AsciiChar* nvinfer1::IPluginV2::getPluginType() const noexcept’
     virtual AsciiChar const* getPluginType() const noexcept = 0;
                              ^~~~~~~~~~~~~
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:205:17: error: looser throw specifier for ‘virtual const char* FlattenConcat::getPluginVersion() const’
     const char* getPluginVersion() const override { return FLATTENCONCAT_PLUGIN_VERSION; }
                 ^~~~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from /home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:437:30: error:   overriding ‘virtual const AsciiChar* nvinfer1::IPluginV2::getPluginVersion() const noexcept’
     virtual AsciiChar const* getPluginVersion() const noexcept = 0;
                              ^~~~~~~~~~~~~~~~
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:207:10: error: looser throw specifier for ‘virtual void FlattenConcat::destroy()’
     void destroy() override {}
          ^~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from /home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:565:18: error:   overriding ‘virtual void nvinfer1::IPluginV2::destroy() noexcept’
     virtual void destroy() noexcept = 0;
                  ^~~~~~~
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:209:16: error: looser throw specifier for ‘virtual nvinfer1::IPluginV2* FlattenConcat::clone() const’
[... 111 lines cut ...]
                  ^~~~~~~~~~~~~~~~~~
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:285:17: error: looser throw specifier for ‘virtual const char* FlattenConcatPluginCreator::getPluginNamespace() const’
     const char* getPluginNamespace() const override
                 ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from /home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:977:30: error:   overriding ‘virtual const AsciiChar* nvinfer1::IPluginCreator::getPluginNamespace() const noexcept’
     virtual AsciiChar const* getPluginNamespace() const noexcept = 0;
                              ^~~~~~~~~~~~~~~~~~
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp: In member function ‘virtual nvinfer1::IPluginV2* FlattenConcatPluginCreator::createPlugin(const char*, const nvinfer1::PluginFieldCollection*)’:
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:271:34: error: invalid new-expression of abstract class type ‘FlattenConcat’
         return new FlattenConcat();
                                  ^
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp: In member function ‘virtual nvinfer1::IPluginV2* FlattenConcatPluginCreator::deserializePlugin(const char*, const void*, size_t)’:
/home/ljx/jetbot/jetbot/ssd_tensorrt/FlattenConcat.cpp:277:58: error: invalid new-expression of abstract class type ‘FlattenConcat’
         return new FlattenConcat(serialData, serialLength);
                                                          ^
jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/build.make:62: recipe for target 'jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/FlattenConcat.cpp.o' failed
make[2]: *** [jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/FlattenConcat.cpp.o] Error 1
CMakeFiles/Makefile2:103: recipe for target 'jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/all' failed
make[1]: *** [jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Unlicensed-driver-ljx, 2022-09-15:
解决了吗

develiberta, 2022-09-29:
How did you solve it?

jiapei100, 2022-10-10:
Same here with **cmake** build.

santaimpersonator, 2023-01-11:
I believe you are following the old instructions from the [wiki](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/software-setup#step-5---install-latest-software-optional). Try using the [latest instructions](https://jetbot.org/master/).