NVIDIA-AI-IOT/jetbot #563: Failed to build jetbot via CMake - error: looser throw specifier for ‘virtual int FlattenConcat::getNbOutputs() const’      int getNbOutputs() const override

jiapei100, 2022-10-10:
```console
[ 50%] Building CXX object jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/FlattenConcat.cpp.o
cd....../build/jetbot/ssd_tensorrt && /usr/bin/c++  -Dssd_tensorrt_EXPORTS -I/usr/local/cuda/include  -O3 -DNDEBUG -fPIC   -o CMakeFiles/ssd_tensorrt.dir/FlattenConcat.cpp.o -c....../jetbot/ssd_tensorrt/FlattenConcat.cpp
....../jetbot/ssd_tensorrt/FlattenConcat.cpp:63:9: error: looser throw specifier for ‘virtual int FlattenConcat::getNbOutputs() const’
     int getNbOutputs() const override
         ^~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from....../jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:469:21: error:   overriding ‘virtual int32_t nvinfer1::IPluginV2::getNbOutputs() const noexcept’
     virtual int32_t getNbOutputs() const noexcept = 0;
                     ^~~~~~~~~~~~
....../jetbot/ssd_tensorrt/FlattenConcat.cpp:69:10: error: looser throw specifier for ‘virtual nvinfer1::Dims FlattenConcat::getOutputDimensions(int, const Dims*, int)’
     Dims getOutputDimensions(int index, const Dims* inputs, int nbInputDims) override
          ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from....../jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:486:18: error:   overriding ‘virtual nvinfer1::Dims nvinfer1::IPluginV2::getOutputDimensions(int32_t, const Dims*, int32_t) noexcept’
     virtual Dims getOutputDimensions(int32_t index, Dims const* inputs, int32_t nbInputDims) noexcept = 0;
                  ^~~~~~~~~~~~~~~~~~~
....../jetbot/ssd_tensorrt/FlattenConcat.cpp:92:9: error: looser throw specifier for ‘virtual int FlattenConcat::initialize()’
     int initialize() override
         ^~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from....../jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:558:21: error:   overriding ‘virtual int32_t nvinfer1::IPluginV2::initialize() noexcept’
     virtual int32_t initialize() noexcept = 0;
                     ^~~~~~~~~~
....../jetbot/ssd_tensorrt/FlattenConcat.cpp:100:10: error: looser throw specifier for ‘virtual void FlattenConcat::terminate()’
     void terminate() override
          ^~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from....../jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:572:18: error:   overriding ‘virtual void nvinfer1::IPluginV2::terminate() noexcept’
     virtual void terminate() noexcept = 0;
                  ^~~~~~~~~
....../jetbot/ssd_tensorrt/FlattenConcat.cpp:107:12: error: looser throw specifier for ‘virtual size_t FlattenConcat::getWorkspaceSize(int) const’
     size_t getWorkspaceSize(int maxBatchSize) const override
            ^~~~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from....../jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:588:20: error:   overriding ‘virtual size_t nvinfer1::IPluginV2::getWorkspaceSize(int32_t) const noexcept’
     virtual size_t getWorkspaceSize(int32_t maxBatchSize) const noexcept = 0;
                    ^~~~~~~~~~~~~~~~
....../jetbot/ssd_tensorrt/FlattenConcat.cpp:113:9: error: ‘int FlattenConcat::enqueue(int, const void* const*, void**, void*, cudaStream_t)’ marked ‘override’, but does not override
     int enqueue(int batchSize, const void* const* inputs, void** outputs, void*, cudaStream_t stream) override
         ^~~~~~~
....../jetbot/ssd_tensorrt/FlattenConcat.cpp:135:12: error: looser throw specifier for ‘virtual size_t FlattenConcat::getSerializationSize() const’
     size_t getSerializationSize() const override
            ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from....../jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:620:20: error:   overriding ‘virtual size_t nvinfer1::IPluginV2::getSerializationSize() const noexcept’
     virtual size_t getSerializationSize() const noexcept = 0;
                    ^~~~~~~~~~~~~~~~~~~~
....../jetbot/ssd_tensorrt/FlattenConcat.cpp:144:10: error: looser throw specifier for ‘virtual void FlattenConcat::serialize(void*) const’
     void serialize(void* buffer) const override
          ^~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferLegacyDims.h:53:0,
[... 173 lines cut ...]
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /usr/include/aarch64-linux-gnu/NvInferPlugin.h:53,
                 from....../jetbot/ssd_tensorrt/FlattenConcat.cpp:9:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:1211:30: error:   overriding ‘virtual const AsciiChar* nvinfer1::IPluginCreator::getPluginNamespace() const noexcept’
     virtual AsciiChar const* getPluginNamespace() const noexcept = 0;
                              ^~~~~~~~~~~~~~~~~~
....../jetbot/ssd_tensorrt/FlattenConcat.cpp: In member function ‘virtual nvinfer1::IPluginV2* FlattenConcatPluginCreator::createPlugin(const char*, const nvinfer1::PluginFieldCollection*)’:
....../jetbot/ssd_tensorrt/FlattenConcat.cpp:271:34: error: invalid new-expression of abstract class type ‘FlattenConcat’
         return new FlattenConcat();
                                  ^
....../jetbot/ssd_tensorrt/FlattenConcat.cpp: In member function ‘virtual nvinfer1::IPluginV2* FlattenConcatPluginCreator::deserializePlugin(const char*, const void*, size_t)’:
....../jetbot/ssd_tensorrt/FlattenConcat.cpp:277:58: error: invalid new-expression of abstract class type ‘FlattenConcat’
         return new FlattenConcat(serialData, serialLength);
                                                          ^
jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/build.make:65: recipe for target 'jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/FlattenConcat.cpp.o' failed
make[2]: *** [jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/FlattenConcat.cpp.o] Error 1
make[2]: Leaving directory '....../build'
CMakeFiles/Makefile2:106: recipe for target 'jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/all' failed
make[1]: *** [jetbot/ssd_tensorrt/CMakeFiles/ssd_tensorrt.dir/all] Error 2
make[1]: Leaving directory '....../build'
Makefile:86: recipe for target 'all' failed
make: *** [all] Error 2
➜  build git:(master) ✗ 

```

lanhong123, 2023-11-11:
解决了吗

jiapei100, 2023-11-11:
@lanhong123  嗯嗯... 我直接略过了... 
视频在这里 [https://www.youtube.com/watch?v=NQCs-tEwwWE](https://www.youtube.com/watch?v=NQCs-tEwwWE)

总之就是：
- 放弃编译 jetbot
- 直接安装jetpack
- 要找到相应的版本，我的是jetson nano，而不是jetsson orin nano，所以jetpack版本一定要找正确了
- 我最想用的好像是jetcam？？

有点忘记了...总之搞了近1个星期...

closed 2023-11-11 by jiapei100

lanhong123, 2023-11-11:
谢谢你的的回答，我刚才找到了原因，我用的是jetsonnao的JetsonNanoUb20_3b.img镜像，想手动添加jetbot库，出现了以上问题，发现是jetbot/ssd_tensorrt/FlattenConcat.cpp里面的一些版本不适配问题导致的，根据问题一条条修改，最后成功安装并运行了Jetbot，修改后的/FlattenConcat.cpp如下：
#include <algorithm>
#include <cassert>
#include <iostream>
#include <numeric>
#include <vector>

#include <cublas_v2.h>
#include "NvInfer.h"
#include "NvInferPlugin.h"

#define CHECK(status)                             \
    do                                            \
    {                                             \
        auto ret = (status);                      \
        if (ret != 0)                             \
        {                                         \
            std::cout << "Cuda failure: " << ret; \
            abort();                              \
        }                                         \
    } while (0)

using namespace nvinfer1;

namespace
{
const char* FLATTENCONCAT_PLUGIN_VERSION{"1"};
const char* FLATTENCONCAT_PLUGIN_NAME{"FlattenConcat_TRT_jetbot"};
}

class FlattenConcat : public IPluginV2
{
public:
    FlattenConcat() {}

    FlattenConcat(const int* flattenedInputSize, int numInputs, int flattenedOutputSize)
        : mFlattenedOutputSize(flattenedOutputSize)
    {
        for (int i = 0; i < numInputs; ++i)
            mFlattenedInputSize.push_back(flattenedInputSize[i]);
    }

    FlattenConcat(const void* data, size_t length)
    {
        const char* d = reinterpret_cast<const char*>(data);
        const char* a = d;

        size_t numInputs = read<size_t>(d);
        for (size_t i = 0; i < numInputs; ++i)
        {
            mFlattenedInputSize.push_back(read<int>(d));
        }
        mFlattenedOutputSize = read<int>(d);

        assert(d == a + length);
    }

    int32_t getNbOutputs() const noexcept override
    {
        return 1;
    }

    Dims getOutputDimensions(int32_t index, const Dims* inputs, int32_t nbInputDims) noexcept override
    {
        assert(nbInputDims >= 1);
        assert(index == 0);

        size_t flattenedOutputSize = 0;
        int32_t inputVolume = 0;

[... 170 lines cut ...]
    }

    IPluginV2* deserializePlugin(const char* name, const void* serialData, size_t serialLength) noexcept override
    {
        return new FlattenConcat(serialData, serialLength);
    }

    void setPluginNamespace(const char* pluginNamespace) noexcept override
    {
        mPluginNamespace = pluginNamespace;
    }

    const char* getPluginNamespace() const noexcept override
    {
        return mPluginNamespace.c_str();
    }

private:
    static PluginFieldCollection mFC;
    std::string mPluginNamespace = "";
};

PluginFieldCollection FlattenConcatPluginCreator::mFC{};

REGISTER_TENSORRT_PLUGIN(FlattenConcatPluginCreator);

lanhong123, 2023-11-11:
`#include <algorithm>
#include <cassert>
#include <iostream>
#include <numeric>
#include <vector>

#include <cublas_v2.h>
#include "NvInfer.h"
#include "NvInferPlugin.h"

#define CHECK(status)                             \
    do                                            \
    {                                             \
        auto ret = (status);                      \
        if (ret != 0)                             \
        {                                         \
            std::cout << "Cuda failure: " << ret; \
            abort();                              \
        }                                         \
    } while (0)

using namespace nvinfer1;

namespace
{
const char* FLATTENCONCAT_PLUGIN_VERSION{"1"};
const char* FLATTENCONCAT_PLUGIN_NAME{"FlattenConcat_TRT_jetbot"};
}

class FlattenConcat : public IPluginV2
{
public:
    FlattenConcat() {}

    FlattenConcat(const int* flattenedInputSize, int numInputs, int flattenedOutputSize)
        : mFlattenedOutputSize(flattenedOutputSize)
    {
        for (int i = 0; i < numInputs; ++i)
            mFlattenedInputSize.push_back(flattenedInputSize[i]);
    }

    FlattenConcat(const void* data, size_t length)
    {
        const char* d = reinterpret_cast<const char*>(data);
        const char* a = d;

        size_t numInputs = read<size_t>(d);
        for (size_t i = 0; i < numInputs; ++i)
        {
            mFlattenedInputSize.push_back(read<int>(d));
        }
        mFlattenedOutputSize = read<int>(d);

        assert(d == a + length);
    }

    int32_t getNbOutputs() const noexcept override
    {
        return 1;
    }

    Dims getOutputDimensions(int32_t index, const Dims* inputs, int32_t nbInputDims) noexcept override
    {
        assert(nbInputDims >= 1);
        assert(index == 0);

        size_t flattenedOutputSize = 0;
        int32_t inputVolume = 0;

        for (int32_t i = 0; i < nbInputDims; ++i)
[... 169 lines cut ...]
    }

    IPluginV2* deserializePlugin(const char* name, const void* serialData, size_t serialLength) noexcept override
    {
        return new FlattenConcat(serialData, serialLength);
    }

    void setPluginNamespace(const char* pluginNamespace) noexcept override
    {
        mPluginNamespace = pluginNamespace;
    }

    const char* getPluginNamespace() const noexcept override
    {
        return mPluginNamespace.c_str();
    }

private:
    static PluginFieldCollection mFC;
    std::string mPluginNamespace = "";
};

PluginFieldCollection FlattenConcatPluginCreator::mFC{};

REGISTER_TENSORRT_PLUGIN(FlattenConcatPluginCreator);`

lanhong123, 2023-11-11:
[Uploading FlattenConcat.txt…]()