NVIDIA-AI-IOT/jetbot #577: jetbot docker images outdated - quickfix

mischa-ai, 2023-02-22:
Obviously the [docker images for jetbot](https://hub.docker.com/r/jetbot/jetbot/tags) received no updates for at least 2 years:

[image]

In the same time the L4T Version has been updated to 32.7.3 (as of Feb 2023).

This leads to some errors when trying to use the docker method for the first time, as the scripts are using the L4T version which is detected automatically in `docker/configure.sh`:

```
export L4T_VERSION="$L4T_RELEASE.$L4T_REVISION"
```

This variable is then used by all `enable.sh` script inside the docker directory.

While simply setting the version to **32.5.0** helps as a quick fix, it doesn't solve the problem of outdated images:

`docker/configure.sh`
```
#!/bin/bash

export JETBOT_VERSION=0.4.3

L4T_VERSION_STRING=$(head -n 1 /etc/nv_tegra_release)
L4T_RELEASE=$(echo $L4T_VERSION_STRING | cut -f 2 -d ' ' | grep -Po '(?<=R)[^;]+')
L4T_REVISION=$(echo $L4T_VERSION_STRING | cut -f 2 -d ',' | grep -Po '(?<=REVISION: )[^;]+')

# comment out this line:
#export L4T_VERSION="$L4T_RELEASE.$L4T_REVISION"
# and add this line to set the available docker images version:
export L4T_VERSION="32.5.0"

...
```

Still one question bothers my mind: is this project abandoned by NVIDIA?

Or simply need some help with a build pipeline to create automatically latest docker images?

mischa-ai, 2023-02-22:
See example: https://github.com/mischa-ai/jetbot/blob/master/docker/configure.sh

SarvagyaVaish, 2024-02-05:
On the docker images corresponding `32.5.0`, the cameras don't seem to initialize. Do you have a workaround for this? Or Is there a way to build updated docker images for the newt L4T versions?