NVIDIA-AI-IOT/jetbot #597: AttributeError: 'NoneType' object has no attribute 'create_execution_context'

jiapei100, 2023-08-03:
```python
from jetbot import ObjectDetector

model = ObjectDetector('ssd_mobilenet_v2_coco.engine')
```

The above code still brought me the following **ERROR** messages:

```console
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-78c468b59bd4> in <module>
      1 from jetbot import ObjectDetector
      2 
----> 3 model = ObjectDetector('ssd_mobilenet_v2_coco.engine')

~/.local/lib/python3.6/site-packages/jetbot/object_detection.py in __init__(self, engine_path, preprocess_fn)
     27         load_plugins()
     28         self.trt_model = TRTModel(engine_path, input_names=[TRT_INPUT_NAME],
---> 29                                   output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
     30         self.preprocess_fn = preprocess_fn
     31 

~/.local/lib/python3.6/site-packages/jetbot/tensorrt_model.py in __init__(self, engine_path, input_names, output_names, final_shapes)
     57         with open(engine_path, 'rb') as f:
     58             self.engine = self.runtime.deserialize_cuda_engine(f.read())
---> 59         self.context = self.engine.create_execution_context()
     60 
     61         if input_names is None:

AttributeError: 'NoneType' object has no attribute 'create_execution_context'
```

Can anybody let me know how to successfully load **ssd_mobilenet_v2_coco.engine** ?? BTW, I'm already using [v0.4 (latest)](https://drive.google.com/open?id=1KjlDMRD8uhgQmQK-nC2CZGHFTbq4qQQH)

sriharish-324, 2023-12-03:
Did you got any solution for this

cccapital0107, 2024-02-12:
The problem is that jetpack 4.5 doesn't support that file.

Can someone please provide the file with jetpack 4.5 support.

ssd_mobilenet_v2_coco.engine

omisha-bajoria, 2024-06-03:
> The problem is that jetpack 4.5 doesn't support that file.
> 
> Can someone provide the file with jetpack 4.5 support?
> 
> ssd_mobilenet_v2_coco.engine

Were you able to find the compatible file or work something else out?