NVIDIA-AI-IOT/jetbot #628: jetbot/jetpack download request ignored?

Tomas61692, 2024-10-01:
im getting no response from jetbot.org when requesting to download the jetbot/jetpack on google groups.

anddopTheGreat, 2024-10-02:
uhhhhhhhh

Tomas61692, 2024-10-02:
> uhhhhhhhh

i take it it shouldlnt have happened?

[image]
[image]

anddopTheGreat, 2024-10-04:
yea i just tried to download it as well and it says permission needed. Are they releasing a new image version because why is the owner of the file not allowing people to download it????

anddopTheGreat, 2024-10-08:
it's fixed now

Tomas61692, 2024-10-10:
@anddopTheGreat 
@jaybdub 
i saw a topic in the jetbot forums where you seem to have figured out how to apply your changes for a 4wd makeover.
i changed a few lines in robot.py as well and folowed issue #372
cd docker
./disable.sh
./configure.sh
./build.sh
./enable.sh

that didnt apply the changes